raspiblitz/raspiblitz #1752: Tor repo doesn't support architecture 'armhf' (32bit ARM) - need to compile from source?

openoms, 2020-11-10:
This is a possible issue for all newly set up Tor nodes:
```
$ sudo apt update
                                     
Hit:5 https://deb.torproject.org/torproject.org buster InRelease                   

N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://deb.torproject.org/torproject.org buster InRelease' doesn't support architecture 'armhf'
```

Reported in RaspiBolt: https://github.com/Stadicus/RaspiBolt/issues/656

If this remains to be an issue there are two obvious solutions:
* switch to a 64bit base image (https://github.com/rootzoll/raspiblitz/issues/1199 - aimed for v1.7)
* compile Tor from source (can help ti get around some issues with some ISP censorship since can be downloaded from github)

Methods to compile Tor from source:

1) 
* https://t.me/RaspiBlitzFAQ/168
* https://tor.stackexchange.com/questions/75/how-can-i-install-tor-from-the-source-code-in-the-git-repository
    ```
    sudo apt-get install -y git build-essential automake libevent-dev libssl-dev zlib1g-dev

    # git clone https://git.torproject.org/tor.git
    git clone https://github.com/torproject/tor

    cd tor
    ./autogen.sh
 
    ./configure --disable-asciidoc 

    make
    make install
    ```

2) 
* https://github.com/openoms/joininbox/blob/4042dde809eb9cba5a3e1c2941dc414eb93251b6/build_joininbox.sh#L343
* https://2019.www.torproject.org/docs/debian#source

   ```
    # https://2019.www.torproject.org/docs/debian#source
    echo "# running on armv6l - need to compile Tor from source"
    apt install -y build-essential fakeroot devscripts
    apt build-dep -y tor deb.torproject.org-keyring
    mkdir ~/debian-packages; cd ~/debian-packages
    apt source tor
    cd tor-*
    debuild -rfakeroot -uc -us
    cd ..
    dpkg -i tor_*.deb
    # setup Tor in the backgound
    # TODO - test if remains in the background after the Tor service is started
    tor &
    ```

rootzoll, 2020-11-11:
I just tested a recover/update with the v1.6.1 image (where Tor gets installed) and it was working - so I pushed this to v1.7 where we anyway consider to test the 64-bit image. @openoms do you see anything pressing for the v1.6.2 release?

frennkie, 2020-11-11:
@rootzoll can you check the version of tor that was installed? I can image that `apt install tor` still work even though https://deb.torproject.org/torproject.org no longer hosts armhf packages because Debian/Raspbian itself also hosts an older version.

I currently have this running:

```
dpkg -l tor
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version                Architecture Description
+++-==============-======================-============-===================================
ii  tor            0.4.2.7-1~d10.buster+1 armhf        anonymizing overlay network for TCP
```

Running `apt-cache show tor` shows that the Duster repo contains 0.3.5.10-1. While this version does include support for security issues (I assume) it will be different regarding features and configurations settings that 0.4+ .

rootzoll, 2020-11-11:
Yes it seems to fallback to a 0.3.5.10 ..
```
admin@raspberrypi:~ $ dpkg -l tor
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version      Architecture Description
+++-==============-============-============-===================================
ii  tor            0.3.5.10-1   armhf        anonymizing overlay network for TCP
```

frennkie, 2020-11-11:
@openoms is there a specific feature/settings that we want/need from tor 0.4.X ? Or could we live with 0.3.5?

openoms, 2020-11-11:
Not sure of the feature differences, but generally it is strongly recommended to run the latest version.

openoms, 2020-11-12:
I've run test compilation with this simple script (from https://2019.www.torproject.org/docs/debian#source):
 ```
 $ cat tor.sh
 apt install -y build-essential fakeroot devscripts
 apt build-dep -y tor deb.torproject.org-keyring
 mkdir ~/debian-packages; cd ~/debian-packages
 apt source tor
 cd tor-*
 debuild -rfakeroot -uc -us
 ```
on a RPi4 4GB:
```
$ sudo time tor.sh 

...

real	18m56.093s
user	36m8.960s
sys	3m45.892s
```

Then upgraded Tor successfully (the node is up and running):
```
root@raspberrypi:~# dpkg -l tor
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version                Architecture Description
+++-==============-======================-============-===================================
ii  tor            0.4.2.7-1~d10.buster+1 armhf        anonymizing overlay network for TCP

root@raspberrypi:~/debian-packages# dpkg -i tor_*.deb
(Reading database ... 102579 files and directories currently installed.)
Preparing to unpack tor_0.4.4.5-1~d10.buster+1_armhf.deb ...
Unpacking tor (0.4.4.5-1~d10.buster+1) over (0.4.2.7-1~d10.buster+1) ...
Setting up tor (0.4.4.5-1~d10.buster+1) ...
Something or somebody made /var/log/tor disappear.
Creating one for you again.
Processing triggers for systemd (241-7~deb10u4+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...

root@raspberrypi:~/debian-packages# dpkg -l tor
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version                Architecture Description
+++-==============-======================-============-===================================
ii  tor            0.4.4.5-1~d10.buster+1 armhf        anonymizing overlay network for TCP
```

Question is if we should take up with the 36 mins build time or maybe should ship the tor_*.debs on the SDcard (compile in the build_sdcard.sh) even if it is not active by default?

openoms, 2020-11-13:
Maybe as Tor 0.3.5.10-1 is working ok on v1.6.1 and I am not aware of any critical issues with running the older version we might keep it for v1.6.2 (so don't need to change the build_sdcard.sh and no long wait on install).

Could offer an optional Tor update in the UPDATE menu which builds from source.

My node had no issues while building and then changing to the new version can be done on the fly too, will just restart Tor afterwards.

Can even even test if it makes any difference if the build is done from the SSD vs the SDcard.

rootzoll, 2020-11-18:
I merged the PR 1769 and for the next v1.6.2 release sd card image I will run once manually the `update` so that the self buil packages will already we on the sd card. Putting on final testing for RC1.

rootzoll, 2020-11-27:
Testing with RC1 & RC2 look fine on the Tor side. So closing this issue for 1.6.2 release.

frennkie, 2021-04-25:
I just installed 1.7 and then ran `sudo apt update` and I get:

```
admin@raspberrypi:~ $ sudo apt update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster InRelease
Hit:3 http://deb.debian.org/debian-security buster/updates InRelease
Hit:4 http://deb.debian.org/debian buster-updates InRelease
Hit:5 https://deb.torproject.org/torproject.org buster InRelease
Reading package lists... Done
Building dependency tree
Reading state information... Done
13 packages can be upgraded. Run 'apt list --upgradable' to see them.
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://deb.torproject.org/torproject.org buster InRelease' doesn't support architecture 'armhf'
```

I am running aarch64:

```
admin@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
```

And Tor is installed as version 0.4.5.7 from the Tor repo:

```
admin@raspberrypi:~ $ dpkg -l tor
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name           Version                Architecture Description
+++-==============-======================-============-===================================
ii  tor            0.4.5.7-1~d10.buster+1 arm64        anonymizing overlay network for TCP
```

@openoms  Any ideas why this warning is still show and how to get rid of it?

frennkie, 2021-04-25:
Adding `[arch=arm64]` seems to help:

```
cat /etc/apt/sources.list
deb http://deb.debian.org/debian buster main contrib non-free
deb http://deb.debian.org/debian-security/ buster/updates main contrib non-free
deb http://deb.debian.org/debian buster-updates main contrib non-free
# Uncomment deb-src lines below then 'apt-get update' to enable 'apt-get source'
#deb-src http://deb.debian.org/debian buster main contrib non-free
#deb-src http://deb.debian.org/debian-security/ buster/updates main contrib non-free
#deb-src http://deb.debian.org/debian buster-updates main contrib non-free

deb [arch=arm64] https://deb.torproject.org/torproject.org buster main
deb-src [arch=arm64] https://deb.torproject.org/torproject.org buster main
```

Then I get no warnings.

```
admin@raspberrypi:~ $ sudo apt update
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian-security buster/updates InRelease
Hit:3 http://deb.debian.org/debian buster-updates InRelease
Hit:4 http://archive.raspberrypi.org/debian buster InRelease
Get:5 https://deb.torproject.org/torproject.org buster InRelease [3,524 B]
Get:6 https://deb.torproject.org/torproject.org buster/main Sources [1,250 B]
Get:7 https://deb.torproject.org/torproject.org buster/main arm64 Packages [2,392 B]
Fetched 7,166 B in 2s (3,530 B/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
```

GoetzGoerisch, 2021-11-20:
On a fresh 1.7.1 image I did:

````
sudo apt-get purge ".*:armhf"
sudo dpkg --remove-architecture armhf
sudo apt update
sudo apt full-upgrade
````

raspiblitz is runing fine so far.

rootzoll, 2021-12-11:
We now use 64bit baseimages by default .. because 32bit are not recommended because of large LND channels.db files etc. Closing issue.

closed 2021-12-11 by rootzoll