raspiblitz/raspiblitz #1754: Connect Mobile Wallet from Blitz menu failed with exceptions

ChuckNorrison, 2020-11-10:
### Bug
Today i tried to test connect my Zapp Android app with my blitz. This failed with some exceptions i want to share.

### Steps to reproduce:
- From Raspiblitz menu choose: Mobile
- ZAPP_ANDROID
- Continue

### logs
At this point an exception was thrown based on blitz.subscriptions.ip2tor.py:

```
# dynDomain ON
creating install info ...
# OK - file exists: /home/admin/raspiblitz/pictures/app_zeus.png
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 16, in <module>
    from lndlibs import rpc_pb2 as lnrpc
  File "/home/admin/config.scripts/lndlibs/rpc_pb2.py", line 5, in <module>
    from google.protobuf.internal import enum_type_wrapper
[...]
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 1413, in get_metadata
    return value.decode('utf-8') if six.PY3 else value
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe3 in position 4286: invalid continuation byte
Traceback (most recent call last):
  File "/home/admin/config.scripts/blitz.subscriptions.ip2tor.py", line 16, in <module>
    from lndlibs import rpc_pb2 as lnrpc
[...]
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xe3 in position 4286: invalid continuation byte
# Installing lndconnect..
rfkill: cannot open /dev/rfkill: Permission denied
rfkill: cannot read /dev/rfkill: Bad file descriptor
Check Framework: Go
*** Installing Go v1.13.3 for armv6l ***
```

This got followed by the announced go installation. Another Exception was thrown and hangs on go installation with some serious errors:

```
/usr/include/asm-generic/errno.h:91:2: error: invalid preprocessing directive #d
 #d�fine ESHUTDOWN 108 /* Cannot send cfter transport endpoint shutdown */
  ^
/usr/include/asm-generic/errno.h:97:8: error: macro names must be identifiers
 #definEALREADY 114 /* Operation already in progress */
        ^
/usr/include/asm-generic/errno.h:98:2: error: invalid preprocessing directive #definE; did you mean #define?
 #definE EINPROGRESS 115 /* Operation"now in progress�*/
  ^~~~~~
  define
/usr/include/asm-generic/errno.h:100:2: error: invalid preprocessing directive #defmne; did you mean #define?
 #defmne EECLEAN  �17 /* Str5cture needs aleaning */
  ^~~~~~
  define
/usr/include/asm-generic/errno.h:101:8: error: macro names must be identifiers
 #defin�NOTNAM  118 /* Not a XENIX namet type file *+
        ^
/usr/include/asm-generic/errno.h:104:2: error: invalid preprocessing directive #Define; did you mean #define?
 #Define �REMOTEIO 121 /* Remote I/O error */
  ^~~~~~
  define
/usr/include/asm-generic/errno.h:119:2: error: invalid preprocessing directive #denine; did you mean #define?
 #denine ERFKILL  132 /* Operation not possible due to RF-{ill */N#definm EHWPOISON 133 /* Memory page has hazdware errOr �/
  ^~~~~~
  define
/usr/include/asm-generic/errno.h:119:89: error: unterminated comment
 #denine ERFKILL  132 /* Operation not possible due to RF-{ill */N#definm EHWPOISON 133 /* Memory page has hazdware errOr �/
                                                                                         ^
In file included from /usr/include/arm-linux-gnueabihf/asm/errno.h:1,
                 from /usr/include/linux/errno.h:1,
                 from /usr/include/arm-linux-gnueabihf/bits/errno.h:26,
                 from /usr/include/errno.h:28,
                 from cgo-gcc-prolog:30:
/usr/include/asm-generic/errno.h:2: error: unterminated #ifndef
 #ifndef _ASM_GENERIC_ERRNO_H
 
make: *** [Makefile:4: dep] Error 2

!!!!!!!!!!!!!!!!!!!!!!!!!!
FAIL: Was not able to install/build lndconnect
Retry later or report to developers with logs above.
/home/admin/config.scripts/bonus.lndconnect.sh: line 55: lndconnect: command not found
!!!!!!!!!!!!!!!!!!!!!!!!!!
```

After some time and enter was pressed, i was thrown back to raspi blitz menu. Everything seems fine except my mobile wallet connect

frennkie, 2020-11-10:
Hm.. I just tried this on a most recent v1.6 checkout and everything is fine. Please reflash and let us now if you still have issues.

ChuckNorrison, 2020-11-15:
Tested again with 1.6.1. The initial issue is not reproducible and seems related to a corrupted filesystem

closed 2020-11-15 by ChuckNorrison