raspiblitz/raspiblitz #1757: Research: Introduce a Plugin Structure

rootzoll, 2020-11-11:
One goal of the RaspiBlitz is to be a platform for the community to test & develop their projects lightning projects on. At the moment we see a growing number of additional apps as "bonus.X.sh" config-scripts. But still evertime other scripts need to get changed on integration too .. like SSH menu, update/recovery, main menu, providing a LCD screen (future), providing a WebUI screen,  etc.

Maybe it makes sense to make this more structured like a plugin system - so additional apps can be better packaged and all code and resources is in one place. With v1.7 we can try a parrallel structure - like a new folder "addons" that then contains one folder per additional app that contains the script and all resources (maybe images, configs, etc needed). A manifest file might not be needed - that can be part of the script so that it even can be dynamic. But those are already details. For v1.7 its more the idea to have a first experimental prototype for one or two "plugins" that can get discussed based on first experience and then gets step by step finanlized towards a 2.0 version.

rootzoll, 2021-03-23:
Pushing this to v1.8 .. plugins can make sense especially when moving to WebUI.

rootzoll, 2021-04-12:
Also take care to describe the dependencies between plugins/apps better.

nyxnor, 2021-04-18:
An option is using the Kixunil repo deb cryptoanarchy. Deb packaging system, just 1 command to install the service, no need for long scripts. https://github.com/rootzoll/raspiblitz/issues/1067

Kixunil, 2021-04-18:
Yes, I already have plugin system and I have refactoring plan to make it very clean. It'll take some time though.