raspiblitz/raspiblitz #1758: [Feature request] SphinX relay

openoms, 2020-11-11:
https://sphinx.chat/

Manual deployment notes (might be outdated):
https://github.com/stakwork/sphinx-relay/blob/master/docs/raspiblitz_deployment.md

rootzoll, 2020-11-11:
Lets give it a shot if we can include this into v1.6.2

rootzoll, 2020-11-12:
OK this is the `app.json` 
```
admin@raspberrypi:/home/sphinxrelay/sphinx-relay $ sudo cat ./config/app.json
{
  "development": {
    "senza_url": "http://localhost:3000/api/v2",
    "macaroon_location": "/Users/evanfeenstra/code/lnd-dev/alice/data/chain/bitcoin/simnet/admin.macaroon",
    "tls_location": "/Users/evanfeenstra/Library/Application Support/Lnd/tls.cert",
    "node_ip": "127.0.0.1",
    "lnd_port": "10001",
    "node_http_protocol": "http",
    "node_http_port": "3002",
    "hub_api_url": "http://lvh.me/api/v1",
    "hub_url": "http://lvh.me/ping",
    "hub_invite_url": "http://lvh.me/invites",
    "hub_check_invite_url": "http://lvh.me/check_invite",
    "media_host": "localhost:5000",
    "tribes_host": "tribes.sphinx.chat"
  },
  "production": {
    "senza_url": "https://staging.senza.us/api/v2/",
    "macaroon_location": "/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/admin.macaroon",
    "tls_location": "/mnt/hdd/app-data/lnd/tls.cert",
    "lnd_log_location": "/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log",
    "node_ip": "localhost",
    "node_http_protocol": "http",
    "node_http_port": "3300",
    "lnd_port": "10009",
    "hub_api_url": "https://hub.sphinx.chat/api/v1",
    "hub_url": "https://hub.sphinx.chat/ping",
    "hub_invite_url": "https://hub.sphinx.chat/invites",
    "hub_check_invite_url": "https://hub.sphinx.chat/check_invite",
    "media_host": "memes.sphinx.chat",
    "tribes_host": "tribes.sphinx.chat",
    "public_url": "http://192.168.178.95:3300"
  }
}
```

So development has port 3002 and production port 3300 ... but when I start it up it tries to open on port 3001 (which is used on my machine) and fails:
```
admin@raspberrypi:/home/sphinxrelay/sphinx-relay $ sudo -u sphinxrelay env NODE_ENV=production /usr/bin/node dist/app.js
=> env: production
=> [db] starting setup...
=> [db] done syncing
=> [db] setup done
=> [lnd] connecting... attempt #1
(node:12274) DeprecationWarning: grpc.load: Use the @grpc/proto-loader module with grpc.loadPackageDefinition instead
(Use `node --trace-deprecation ...` to show where the warning was created)
events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE: address already in use :::3001
    at Server.setupListenHandle [as _listen2] (net.js:1318:16)
    at listenInCluster (net.js:1366:12)
    at Server.listen (net.js:1452:7)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:90:16
    at Generator.next (<anonymous>)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/home/sphinxrelay/sphinx-relay/dist/app.js:4:12)
    at setupApp (/home/sphinxrelay/sphinx-relay/dist/app.js:74:12)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:69:15
    at Generator.next (<anonymous>)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/home/sphinxrelay/sphinx-relay/dist/app.js:4:12)
    at mainSetup (/home/sphinxrelay/sphinx-relay/dist/app.js:64:12)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:47:19
    at Generator.next (<anonymous>)
    at fulfilled (/home/sphinxrelay/sphinx-relay/dist/app.js:5:58)
Emitted 'error' event on Server instance at:
    at emitErrorNT (net.js:1345:8)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  code: 'EADDRINUSE',
  errno: -98,
  syscall: 'listen',
  address: '::',
  port: 3001
}
```

rootzoll, 2020-11-12:
```
admin@raspberrypi:/home/sphinxrelay/sphinx-relay $ sudo cat ./dist/config/app.json
{
  "development": {
    "senza_url": "http://localhost:3000/api/v2",
    "macaroon_location": "/Users/evanfeenstra/code/lnd-dev/alice/data/chain/bitcoin/simnet/admin.macaroon",
    "tls_location": "/Users/evanfeenstra/Library/Application Support/Lnd/tls.cert",
    "node_ip": "127.0.0.1",
    "lnd_port": "10001",
    "node_http_protocol": "http",
    "node_http_port": "3002",
    "hub_api_url": "http://lvh.me/api/v1",
    "hub_url": "http://lvh.me/ping",
    "hub_invite_url": "http://lvh.me/invites",
    "hub_check_invite_url": "http://lvh.me/check_invite",
    "media_host": "localhost:5000",
    "tribes_host": "tribes.sphinx.chat"
  },
  "production": {
    "senza_url": "https://staging.senza.us/api/v2/",
    "macaroon_location": "/mnt/hdd/app-data/lnd/data/chain/bitcoin/mainnet/admin.macaroon",
    "tls_location": "/mnt/hdd/app-data/lnd/tls.cert",
    "lnd_log_location": "/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log",
    "node_ip": "localhost",
    "node_http_protocol": "http",
    "node_http_port": "3300",
    "lnd_port": "10009",
    "hub_api_url": "https://hub.sphinx.chat/api/v1",
    "hub_url": "https://hub.sphinx.chat/ping",
    "hub_invite_url": "https://hub.sphinx.chat/invites",
    "hub_check_invite_url": "https://hub.sphinx.chat/check_invite",
    "media_host": "memes.sphinx.chat",
    "tribes_host": "tribes.sphinx.chat",
    "public_url": "http://192.168.178.95:3300"
  }
}
```

```
admin@raspberrypi:/home/sphinxrelay/sphinx-relay $ sudo -u sphinxrelay env NODE_ENV=production /usr/bin/node dist/app.js
=> env: production
=> [db] starting setup...
=> [db] done syncing
=> [db] setup done
=> [lnd] connecting... attempt #1
(node:11097) DeprecationWarning: grpc.load: Use the @grpc/proto-loader module with grpc.loadPackageDefinition instead
(Use `node --trace-deprecation ...` to show where the warning was created)
events.js:292
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE: address already in use :::3001
    at Server.setupListenHandle [as _listen2] (net.js:1318:16)
    at listenInCluster (net.js:1366:12)
    at Server.listen (net.js:1452:7)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:90:16
    at Generator.next (<anonymous>)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/home/sphinxrelay/sphinx-relay/dist/app.js:4:12)
    at setupApp (/home/sphinxrelay/sphinx-relay/dist/app.js:74:12)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:69:15
    at Generator.next (<anonymous>)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/home/sphinxrelay/sphinx-relay/dist/app.js:4:12)
    at mainSetup (/home/sphinxrelay/sphinx-relay/dist/app.js:64:12)
    at /home/sphinxrelay/sphinx-relay/dist/app.js:47:19
    at Generator.next (<anonymous>)
    at fulfilled (/home/sphinxrelay/sphinx-relay/dist/app.js:5:58)
Emitted 'error' event on Server instance at:
    at emitErrorNT (net.js:1345:8)
    at processTicksAndRejections (internal/process/task_queues.js:80:21) {
  code: 'EADDRINUSE',
  errno: -98,
  syscall: 'listen',
  address: '::',
  port: 3001
}
```

rootzoll, 2020-11-12:
OK I got the SphinxRelay working in the local network and I prepared now to also have it as an option for IP2TOR.

rootzoll, 2020-11-13:
Tested recovery/update. Next todos:
- [ ] finish IP2Tor option
- [ ] add to "MobileApp" with instructions and offering of IP2Tor
- [ ] get pairing code and display as QR on LCD
- [ ] adapt main menu with hint mobile menu

rootzoll, 2020-11-16:
TODO: Test IP2TOR with iOS SphinxApp

rootzoll, 2020-11-17:
OK Sphinx Relay Feature should now be ready to test. Either with a v1.6.1 release go to MAINMENU > UPDATE > PATCH > PATCH-MENU > PATCH & reboot -> then you should find the "Sphinx Relay" option under MAINMENU > SERICES ... or wait for the first release candidate for the v1.6.2 coming up the next days.

dennisreimann, 2020-11-17:
Awesome – that worked! 👍

Feelin' #reckless again 🚀

rootzoll, 2020-11-18:
OK did some improvements for IP2TOR setup ... works fine on my machine. Will keep issue open for one last setup test when RC1 is out.

rootzoll, 2020-11-18:
RaspiBlitz v1.6.2RC1 is out for testing: https://github.com/rootzoll/raspiblitz/issues/1785

Lace79, 2020-11-19:
Installed First way:
`Go to MAINMENU > UPDATE > PATCH > PATCH-MENU > PATCH & reboot ... you need todo this two times because the patch script itself gets updates the first time. -> then you should find the "Sphinx Relay" option under MAINMENU > SERICES.`

I don't get a QR Code to scan? Display and Terminal - no thin to scan ? 

Error on Terminal:  `Failed to encode the input data: Invalid argument`

rootzoll, 2020-11-19:
@Lace79 can you go to the terminal and run `/home/admin/config.scripts/bonus.sphinxrelay.sh status` .. has the `connectionCode` a value?

Lace79, 2020-11-19:
`connectionCode=''`
it seems it has now value

rootzoll, 2020-11-19:
Sometime it takes some time. If the connection code has no or outdated value it should give you a warning when you open `SPHINX` in main menu. Did you saw that warning?

Lace79, 2020-11-19:
Yes i saw the warning - it is not locally responding and not over tor. how long does it take?

rootzoll, 2020-11-19:
From the `status` output test in your browser: http://[localIP]:3300 ... it should answere is a `Cannot GET /` then its running locally OK.

rootzoll, 2020-11-23:
With PR #1796 I made the warning message stronger when Sphinx is not ready yet.

openoms, 2020-11-26:
https://github.com/stakwork/sphinx-relay/releases
new version v1.1.1
Would it worth it to test and maybe include an update option too?

rootzoll, 2020-11-26:
Will update the version and quick test. If you like to add an "update option" within th next day that would be great ... I target to package 1.6.2 final (if nothing critical shows up) tomorrow.

rootzoll, 2020-11-26:
OK updated the version & added an update option. Ready for merge & final testing.

openoms, 2020-11-27:
got a permission error see PR: https://github.com/rootzoll/raspiblitz/pull/1809

openoms, 2020-11-27:
also I followed the update method in https://github.com/stakwork/sphinx-relay/blob/master/docs/raspiblitz_deployment.md#fast-method beacuse the Thub method would fail to fetch and the overwrite the config (since it is in the sphinx-relay dir)

rootzoll, 2020-11-27:
@openoms ah OK the config overwrite could be a think still on my impelementation. Please rebase again against my latest merge.

rootzoll, 2020-11-27:
@openoms to preserve the config with the git stash
 does not seem to work - see my test results:
```
```

rootzoll, 2020-11-27:
Maybe a better and safer solution woulds be also to rewrite the `config.json` on every start with the `write-environment` part of the script ... let me try this out.

rootzoll, 2020-11-27:
OK this looks good now on my testing :) config gets updated on every start now - so we can just update the code and it will be rewritten on restart.

openoms, 2020-11-27:
found only one last thing when testing the update: https://github.com/rootzoll/raspiblitz/pull/1812

openoms, 2020-11-28:
Good to have the update option, the releases are very frequent: https://github.com/stakwork/sphinx-relay/releases/tag/v1.1.3

rootzoll, 2020-11-28:
Update v1.1.3 tested - will be part of v1.6.2 release. Closing issue.

closed 2020-11-28 by rootzoll