raspiblitz/raspiblitz #1762: Update LNBalances Script

rootzoll, 2020-11-11:
This maybe a nice upgrade for the `lnbalance` script of the RaspiBlitz.
https://twitter.com/gcomxx/status/1325720963258994691?s=20

rootzoll, 2020-11-11:
PR merged. TODO: Test after fresh install or recovery.

rootzoll, 2020-11-12:
When I tested the script on a fresh node (zero balance but one channel was opned to that node) I get this error:
[image]

gcomte, 2020-11-13:
Will take care of this this weekend.

gcomte, 2020-11-13:
Fixed it with [this commit](https://github.com/gcomte/ln-gems/commit/2c5fe492fd705d7b04b46bb62549144c9a120082) but it looks like there are some more issues

gcomte, 2020-11-13:
FYI: https://github.com/gcomte/ln-gems/issues/3

rootzoll, 2020-11-17:
OK cool. Tested and looking good. Will be part of v1.6.2 release. Closing issue.

closed 2020-11-17 by rootzoll