raspiblitz/raspiblitz #1778: [Feature request] Compact LND database

openoms, 2020-11-17:
Now the second time I had repeated LND restarts after the channel.db grew above 1GB.
Solved with compacting as documented here:
https://openoms.gitbook.io/lightning-node-management/lnddatabasecompaction

[image]

Check the size of the channel.db on the blitz:  
`sudo du -h /mnt/hdd/lnd/data/graph/mainnet/channel.db`

The script could have a good place in the REPAIR menu to compact on demand rather then automatically.

Curious if this issue will be solved with [moving to 64bit](https://github.com/rootzoll/raspiblitz/issues/1199) as well. 

Anyway the compaction has a performance benefit as well so will be a good option to have.

rootzoll, 2020-11-18:
If there a PR incomming or to be pushed to v1.7?

openoms, 2020-11-18:
Not so urgent I think since only I had the issue so far AFAIK. In any case can point people to the guide already. Need to check if the next major LND version will have the compaction built in as well.

openoms, 2020-11-18:
Ok, it seems it is coming in LND v0.12 amd already in the master so don't need to  implement this with chantools: https://github.com/lightningnetwork/lnd/issues/4131

Will just leave this issue here until the release.

openoms, 2021-03-18:
This has been tested and solved with moving to a 64bit base image.

closed 2021-03-18 by openoms