raspiblitz/raspiblitz #1783: was trying to install something using menu...

DrBlackross, 2020-11-18:
this was just the apt-get run, trying to see if it was something simple. but my scroll back buffer doesn't go far enough back to when i was installing service mempool, 

RTL already broke lol (thunderhub works)

going to reinstall mempool and set a log file up, and will post it here when its done.

admin@raspberrypi:~ $ sudo apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 https://deb.torproject.org/torproject.org buster InRelease
Reading package lists... Done
N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'https://deb.torproject.org/torproject.org buster InRelease' doesn't support architecture 'armhf

DrBlackross, 2020-11-18:
it has to do with armhf mariadb packages.... but reinstalling it again. 

but towards the end
ERROR 1008 (HY000) at line 1: Can't drop database 'mempool'; database doesn't exist

heres the errors

[         .........] / extract:http-errors: sill tarball trying to-fast-properties@1.0.3 by hash: sha1-uDVx+k2MJbguIxsG46MFXeTKGkc=

[error] Error: Cannot find module '@angular-devkit/build-angular/package.json'
Require stack:
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/node_modules/@angular-devkit/architect/node/node-modules-architect-host.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/node_modules/@angular-devkit/architect/node/index.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/models/architect-command.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/commands/build-impl.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/node_modules/@angular-devkit/schematics/tools/export-ref.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/node_modules/@angular-devkit/schematics/tools/index.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/utilities/json-schema.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/models/command-runner.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/lib/cli/index.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/lib/init.js
- /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/bin/ng
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:957:15)
    at Function.resolve (internal/modules/cjs/helpers.js:83:19)
    at WorkspaceNodeModulesArchitectHost.resolveBuilder (/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/node_modules/@angular
-devkit/architect/node/node-modules-architect-host.js:42:41)
    at BuildCommand.initialize (/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/models/architect-command.js:139:55)
    at async BuildCommand.validateAndRun (/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/models/command.js:124:9)
    at async Object.runCommand (/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/models/command-runner.js:204:24)
    at async default_1 (/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/@angular/cli/lib/cli/index.js:80:31)

and 

0 info it worked if it ends with ok
1 verbose cli [
1 verbose cli   '/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/bin/node',
1 verbose cli   '/usr/bin/npm',
1 verbose cli   'run',
1 verbose cli   'build'
1 verbose cli ]
2 info using npm@6.14.4
3 info using node@v12.16.3
4 verbose run-script [ 'prebuild', 'build', 'postbuild' ]
5 info lifecycle mempool-frontend@1.0.0~prebuild: mempool-frontend@1.0.0
6 info lifecycle mempool-frontend@1.0.0~build: mempool-frontend@1.0.0
7 verbose lifecycle mempool-frontend@1.0.0~build: unsafe-perm in lifecycle true
8 verbose lifecycle mempool-frontend@1.0.0~build: PATH: /usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/npm/node_modules/npm-lifecycle/node
-gyp-bin:/home/mempool/mempool/frontend/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
9 verbose lifecycle mempool-frontend@1.0.0~build: CWD: /home/mempool/mempool/frontend
10 silly lifecycle mempool-frontend@1.0.0~build: Args: [ '-c', 'ng build --prod' ]
11 info lifecycle mempool-frontend@1.0.0~build: Failed to exec build script
12 verbose stack Error: mempool-frontend@1.0.0 build: `ng build --prod`
12 verbose stack spawn ENOENT
12 verbose stack     at ChildProcess.<anonymous> (/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.
js:48:18)
12 verbose stack     at ChildProcess.emit (events.js:310:20)
12 verbose stack     at maybeClose (internal/child_process.js:1021:16)
12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
13 verbose pkgid mempool-frontend@1.0.0
14 verbose cwd /home/mempool/mempool/frontend
15 verbose Linux 4.19.118-v7l+
16 verbose argv "/usr/local/lib/nodejs/node-v12.16.3-linux-armv7l/bin/node" "/usr/bin/npm" "run" "build"
17 verbose node v12.16.3
18 verbose npm  v6.14.4
19 error code ELIFECYCLE
20 error syscall spawn
21 error file sh
22 error errno ENOENT
23 error mempool-frontend@1.0.0 build: `ng build --prod`
23 error spawn ENOENT
24 error Failed at the mempool-frontend@1.0.0 build script.
24 error This is probably not a problem with npm. There is likely additional logging output above.
25 verbose exit [ 1, true ]

then once it finishes drops back to the menu and clears the scrollback buffer in putty, thats about all i can pull out of the putty log

rootzoll, 2020-11-18:
The first one could be related to #1752

Which version of RaspiBlitz are you running?

bitcoinheiro, 2020-11-22:
Same issue, cannot install any service. All fail. Latest version 1.6.1

DrBlackross, 2020-11-22:
Yep same version

DrBlackross, 2020-11-22:
Sorry been out of state and in the mountains, I'll be home tonight to smash
the keyboard and yell.

;)

rootzoll, 2020-11-23:
Your problem should be (at least party) addressed with the upcoming v1.6.2 release. If you like you can already test the first release candidate - details RC1 here: https://github.com/rootzoll/raspiblitz/issues/1785

DrBlackross, 2020-11-29:
k cool, sorry got stuck out of state a bit tooooo long lol

DrBlackross, 2020-12-10:
ok updated, tried to fire up RTL didn't work, tried to install it (since it wasn't there in the menu). did uninstall reboot then reinstall reboot with RTL etc...... nope

snap of the install screen

OK - RTL code copy looks good

*** Run: npm install ***
npm ERR! cb() never called!

npm ERR! This is an error with npm itself. Please report this error at:
npm ERR!     <https://npm.community>

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/rtl/.npm/_logs/2020-12-10T01_09_55_838Z-debug.log

(also if you close the menu screen the back buffer is gone)

THE LOG lol
admin@raspberrypi:/home/rtl/.npm/_logs $ more /home/admin/2020-12-10T01_09_55_838Z-debug.log
0 info it worked if it ends with ok
1 verbose cli [
1 verbose cli   '/usr/local/lib/nodejs/node-v14.15.0-linux-armv7l/bin/node',
1 verbose cli   '/usr/bin/npm',
1 verbose cli   'install',
1 verbose cli   '--only=prod'
1 verbose cli ]
2 info using npm@6.14.8
3 info using node@v14.15.0
4 verbose npm-session a0449bc21f23e860
5 silly install runPreinstallTopLevelLifecycles
6 silly preinstall rtl@0.9.3-beta
7 info lifecycle rtl@0.9.3-beta~preinstall: rtl@0.9.3-beta
8 silly install loadCurrentTree
9 silly install readLocalPackageData
10 timing stage:loadCurrentTree Completed in 45ms
11 silly install loadIdealTree
12 silly install cloneCurrentTreeToIdealTree
13 timing stage:loadIdealTree:cloneCurrentTree Completed in 1ms
14 silly install loadShrinkwrap
15 timing stage:loadIdealTree:loadShrinkwrap Completed in 4210ms
16 silly install loadAllDepsIntoIdealTree
17 timing stage:loadIdealTree:loadAllDepsIntoIdealTree Completed in 6092ms
18 timing stage:loadIdealTree Completed in 12603ms
19 silly currentTree rtl@0.9.3-beta
20 silly idealTree rtl@0.9.3-beta
20 silly idealTree ├─┬ @angular-devkit/architect@0.901.10
20 silly idealTree │ └── rxjs@6.5.4
20 silly idealTree ├─┬ @angular-devkit/build-angular@0.901.10
20 silly idealTree │ ├── ajv@6.12.0
20 silly idealTree │ ├── parse5@4.0.0
20 silly idealTree │ ├── rxjs@6.5.4
20 silly idealTree │ ├── semver@7.1.3
20 silly idealTree │ └── source-map@0.7.3
20 silly idealTree ├─┬ @angular-devkit/build-optimizer@0.901.10
20 silly idealTree │ ├── source-map@0.7.3
20 silly idealTree │ ├── tslib@1.11.1
20 silly idealTree │ └── typescript@3.6.5
20 silly idealTree ├─┬ @angular-devkit/build-webpack@0.901.10
20 silly idealTree │ └── rxjs@6.5.4
20 silly idealTree ├─┬ @angular-devkit/core@9.1.10
20 silly idealTree │ ├── ajv@6.12.0
20 silly idealTree │ ├── rxjs@6.5.4
20 silly idealTree │ └── source-map@0.7.3
20 silly idealTree ├─┬ @angular-devkit/schematics@9.1.10
20 silly idealTree │ └── rxjs@6.5.4
20 silly idealTree ├── @angular/animations@9.1.12
20 silly idealTree ├── @angular/cdk@9.2.4
20 silly idealTree ├─┬ @angular/cli@9.1.10
20 silly idealTree │ ├── ansi-colors@4.1.1
20 silly idealTree │ ├── debug@4.1.1
[... 1522 lines cut ...]
20 silly idealTree ├── y18n@4.0.0
20 silly idealTree ├── yallist@4.0.0
20 silly idealTree ├── yargs-parser@18.1.3
20 silly idealTree ├── yargs@15.3.0
20 silly idealTree ├── yeast@0.1.2
20 silly idealTree ├── yn@3.1.1
20 silly idealTree └── zone.js@0.10.3
21 silly install generateActionsToTake
22 silly diff-trees filtering actions: includeDev false includeProd true includeOpt true
23 timing stage:generateActionsToTake Completed in 601ms
24 silly diffTrees action count 238
25 silly diffTrees add tslib@2.0.0

~~~(too long to paste the log and couldnt download through SCP don't know why)~~~

2885 silly extract @angular/core@9.1.12 extracted to /home/rtl/RTL/node_modules/.staging/@angular/core-8ac3641f (361308ms)
2886 silly extract rxjs@6.5.5 extracted to /home/rtl/RTL/node_modules/.staging/rxjs-23d43693 (392744ms)
2887 silly extract @angular/common@9.1.12 extracted to /home/rtl/RTL/node_modules/.staging/@angular/common-f2fc25b3 (394888ms)
2888 silly extract typescript@3.8.3 extracted to /home/rtl/RTL/node_modules/.staging/typescript-f5a1a9be (423129ms)
2889 silly extract @angular/material@9.2.4 extracted to /home/rtl/RTL/node_modules/.staging/@angular/material-d651ff7c (454473ms)
2890 timing npm Completed in 715878ms
2891 error cb() never called!
2892 error This is an error with npm itself. Please report this error at:
2893 error <https://npm.community>
EOF

DrBlackross, 2020-12-10:
the node overall is still running...
https://1ml.com/node/0282b3650afa7a0853231c5b2aed5da16a969730117541c9896fb7b9ce7d9ba306/channels?order=closedchannels

thunderhub works, but i don't think the auto pilot and auto balance is working

DrBlackross, 2020-12-10:
hey would it be ok to update node to a higher version
(nvm is god)

openoms, 2020-12-10:
starting with a new, clean SDcard helps sorting out the result of customs steps you took. The wallets and other personal data is restored from the disk and you will have a new clean OS in a tested state. Would recommend to upgrade this way to v1.6.2

DrBlackross, 2020-12-24:
yea just got back in from out of country (again), and before i left i did a full reflash on a sdcard with reinstall... it did keep all my channels and everything, but i cant uninstall or reinstall RTL from the menu still.

also i can't change my peers or my channels (or really any settings) on the Thunderhub web page... to me i think it might be the permissions of the directories on the ssd where everything is stored maybe?

DrBlackross, 2020-12-24:
(btw i am running 1.6.2)

DrBlackross, 2021-01-08:
trying again
with a super fresh install of raspiblitz

DrBlackross, 2021-01-08:
Figures, with BRAND new sdcard, it recovered my backup from the usb thumb drive and Finally got RTL to work with a fresh install but i used the Update menu option, let it shut everything down.. stuck the new NEW freshly imaged sdcard in powered the node back up and BOOM everything works (after redoing all the passwords and stuff, then reinstalling RTL from the services menu, yada, yada...)

now on to autopilot..........

DrBlackross, 2021-01-08:
n/m not touching it

nyxnor, 2021-04-18:
@DrBlackross How is it going, running smoothly now? v1.7 is coming, probably in less than a week. If this issue was solved, pls close this.

nyxnor, 2021-05-17:
@rootzoll Please close this.

closed 2021-05-17 by rootzoll