raspiblitz/raspiblitz #1784: RC1 when installing mempool script needs user input

rootzoll, 2020-11-18:
[image]

This could block during recover phase. Check how this was fixed before.

ChuckNorrison, 2020-11-18:
Best answer is to set environment var for this:
`export NG_CLI_ANALYTICS=ci`

rootzoll, 2020-11-23:
@ChuckNorrison hmmm ... that is already in the code:
https://github.com/rootzoll/raspiblitz/blob/0c5b549fb042854daa99958df81f65a3e816ca21/home.admin/config.scripts/bonus.angular_cli.sh#L46

rootzoll, 2020-11-23:
I will add the environment var also to the `.bashrc` of the admin.

rootzoll, 2020-11-24:
Ok tested again that mempool is running thru on recovery without blocking. Last test todo on a fresh install if that interaction is still showing up.

rootzoll, 2020-11-24:
OK last PR shows no working test. Closing issue for v1.6.2 release.

closed 2020-11-24 by rootzoll