raspiblitz/raspiblitz #1785: Testing v1.6.2 Release Candidate 1

rootzoll, 2020-11-18:
While working on the updated dcumentation here is the RC1 of v1.6.2 to test:
https://raspiblitz.org/raspiblitz-v1.6.2rc1-2020-11-18.img.gz
SHA-256: cc98dfe1cd8071e30d7db9c0639f8d04eae2be5ec274a177b8a39924c653dd77

Focus for testing:
- genral install & update
- LNbits running behind IP2Tor+LetsEncrypt hosting Vouchers & TPoS
- Sphinx-Relay install & connecting app thru IP2Tor+LetsEncrypt
- Lightning Pool
- improved BTRFS (with Raid)

You can write your reports here and I will then open new issues or update existing ones if needed.

holgern, 2020-11-18:
I have BTRFS with RAID and the update worked great. I have also electrs and specter installed.
Great work!

The34Goon, 2020-11-19:
Went from 1.6.0 to 1.6.2RC1
now LND is stuck in a loop.
/dev/fd/58: line 1: syntax error near unexpected token `('
/dev/fd/58: line 1: `Binary file (standard input) matches'

Went and disabled Tor in the menu. Now the menu wont start at all.

-- Logs begin at Thu 2020-11-19 14:32:02 GMT. --
32 RESTARTS DETECTED - LND might be in a error loop
Nov 19 15:05:21 raspberrypi lnd[22212]: /home/bitcoin/.lnd/lnd.conf:31: unknown option: tor.active

Emzy, 2020-11-19:
First ssh login works (It is installing/compiling suff)
Second ssh longin is not working bcause ssh is not running.
Logging in on console and stating ssh ```service ssh start``` fixes it.

I'm using a HDD. It seems it is not mounted at the time of the ssh start. 
/var/log/syslog
```
Nov 19 16:02:07 raspberrypi sshd[744]: /etc/ssh/sshd_config: No such file or directory
.....
Nov 19 16:02:09 raspberrypi tor[1222]: Nov 19 16:02:09.754 [warn] Error creating directory /mnt/hdd/tor/sys: No such file or directory
```

Emzy, 2020-11-19:
SphinxRelay is not starting
syslog:
```
Nov 19 17:48:48 raspberrypi env[30694]: SyntaxError: /home/sphinxrelay/sphinx-relay/dist/config/config.json: Unexpected end of JSO
N input
```

The file /home/sphinxrelay/sphinx-relay/dist/config/config.json was 0 bytes.

Copying the original back und doing 
```
    sudo cat /home/sphinxrelay/sphinx-relay/config/config.json | \
    jq ".production.storage = \"/mnt/hdd/app-data/sphinxrelay/sphinx.db\"" | \
    sudo -u sphinxrelay tee /home/sphinxrelay/sphinx-relay/config/config.json
```
fixed it. 
But this is exactly the part out of the /home/admin/config.scripts/bonus.sphinxrelay.sh
No idea why it was not working on the first run.

grnqrtr, 2020-11-19:
The only problem I've found so far is that kindle display stopped working.
On Kindle display and in browser at `RASPI_IP:3030` all I see is: `Could not open or parse data.json`

```
kindledisplay@raspberrypi:~ $ /home/kindledisplay/kindle-display/server/cron.sh 
Tor is required, but not available.
*** You are running in headless mode.
  Recompressing IDAT chunks in screenshot.png to display.png
   Total length of data found in critical chunks            =      8240
   Best pngcrush method        =   7 (ws 15 fm 0 zl 9 zs 0) =      3265
CPU time decode 0.102192, encode 0.109192, other 0.011934, total 0.254578 sec
```

It says `Tor is required, but not available.` but Tor seems to be running:
```
admin@raspberrypi:~ $ sudo systemctl status tor
● tor.service - Anonymizing overlay network for TCP (multi-instance-master)
   Loaded: loaded (/lib/systemd/system/tor.service; enabled; vendor preset: enabled)
   Active: active (exited) since Thu 2020-11-19 21:54:36 GMT; 1h 43min ago
 Main PID: 456 (code=exited, status=0/SUCCESS)
    Tasks: 0 (limit: 4915)
   Memory: 0B
   CGroup: /system.slice/tor.service

Nov 19 21:54:35 raspberrypi systemd[1]: Starting Anonymizing overlay network for TCP (multi-instance-master)...
Nov 19 21:54:36 raspberrypi systemd[1]: Started Anonymizing overlay network for TCP (multi-instance-master).

```
Everything else, including Sphinx seems to be working fine for me!

holgern, 2020-11-20:
@grnqrtr There is a port conflict. The kindle display is using port 3030 as well as sphinx.

grnqrtr, 2020-11-20:
I just changed `SERVER_PORT=` to `3040` in `config.scripts/bonus.kindle-display.sh` but still get `Could not open or parse data.json` at  `RASPI_IP:3040` after reinstall.

holgern, 2020-11-20:
The port seems to be hardcoded to 3030:

https://github.com/dennisreimann/kindle-display/blob/master/kindle/mnt/base-us/update.sh

holgern, 2020-11-20:
@grnqrtr You need to change the port back to 3030
and install the following packages:
```
sudo apt-get update
sudo apt-get install firefox-esr jo torsocks pngcrush
```
When asked, keep the current tor setup (N).

holgern, 2020-11-20:
@grnqrtr Can you test if PR #1790 fixes your issue?

grnqrtr, 2020-11-20:
@holgern Yes, this fixed it for me. Thanks!

rootzoll, 2020-11-23:
Thanks giving the RC1 a test ride. I have seperated all reported errors into single issues if neede so far. Preparing a RC2 this week.

rootzoll, 2020-11-26:
--- RaspiBlitz v1.6.2 RC2 DOWNLOAD ----
If you have a chance please give this Release Candidate 2 a quick test ride and let me know if you still find anything:
https://raspiblitz.org/raspiblitz-v1.6.2rc2-2020-11-25.img.gz
SHA256: 096cbf03293c6691caa620db14806543ba7200b52c7ee0350fbb4dc5586485f3

jodobear, 2020-11-29:
UPDATE on 1.6.2 RC2 with fresh sd card image, fresh USB3 & unformatted ssd(never reached format options stage):

- On startup `data mount failed` on top line where blitz displays `Raspiblitz` and version.
- On ssh i get `(standard_in) 1: syntax error` while loading.
- On selecting to go ahead with re-setup. After Bitcoin > Hostname > Pass A > Pass B > Tor i get the following error:
```sh
HDD was already initialized/prepared
Now needs to be mounted

# *** 40addHDD.sh ***
(standard_in) 1: syntax error
mount: /mnt/hdd: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error.

# --> Linking drives/directories
# hddCandidate='sda'
# hddPartitionCandidate='sda2'
FAIL blitz.datadrive.sh link --> no data drive mounted
```

rootzoll, 2020-11-29:
@jodobear this looks like the problem I am working in. After the script was optimized for VMs it was s not detecting HDDs without any partions anymore.

rootzoll, 2020-12-02:
OK all test look good. Close for v1.6.2 release.

closed 2020-12-02 by rootzoll

jodobear, 2020-12-04:
I tried 1.6.2, but still getting the same error :

> UPDATE on 1.6.2 RC2 with fresh sd card image, fresh USB3 & unformatted ssd(never reached format options stage):
> 
> * On startup `data mount failed` on top line where blitz displays `Raspiblitz` and version.
> * On ssh i get `(standard_in) 1: syntax error` while loading.
> * On selecting to go ahead with re-setup. After Bitcoin > Hostname > Pass A > Pass B > Tor i get the following error:
> 
> ```shell
> HDD was already initialized/prepared
> Now needs to be mounted
> 
> # *** 40addHDD.sh ***
> (standard_in) 1: syntax error
> mount: /mnt/hdd: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error.
> 
> # --> Linking drives/directories
> # hddCandidate='sda'
> # hddPartitionCandidate='sda2'
> FAIL blitz.datadrive.sh link --> no data drive mounted
> ```