raspiblitz/raspiblitz #1791: Update: Thunderhub 0.10.4

rootzoll, 2020-11-20:
with lnmarkets & boltz integration

rootzoll, 2020-11-20:
PR merged and first test worked. Final testing is needed for recovery with RC2 release.

rootzoll, 2020-11-24:
Recovery runs fine. Closing issue for v1.6.2 release.

kilrau, 2020-11-25:
Failed to manually build an image from master yesterday with the script, still checking what went wrong. In the meanwhile - do you have a RC2 image ready somewhere? @rootzoll

rootzoll, 2020-11-25:
Reopen for final testing. RC2 is coming today or tomorrow.

rootzoll, 2020-11-25:
I just tested the thunderhub install again with the latest script and it worked - so i will pack it up for RC2 for more community testing. If any install error occurs any error message details help.

rootzoll, 2020-11-26:
v1.6.2RC2 is available - see: https://github.com/rootzoll/raspiblitz/issues/1785#issuecomment-734025137

kilrau, 2020-11-27:
Couldn't find any issues.

rootzoll, 2020-11-27:
OK closing issue for v1.6.2 release.

openoms, 2020-11-29:
The ThunderHub  service should be `torifiy`-d same as it i done with Pool: https://github.com/rootzoll/raspiblitz/blob/077ee8b6ad82d78a69c2f4ae7e493ba5dfbbfcfd/home.admin/config.scripts/bonus.pool.sh#L94

(and Loop has native proxy support)

rootzoll, 2020-11-29:
@openoms can we push the torifying of services to 1.7 where we have the milestone to go Tor by default. I want to build the final v1.6.2 release today?

openoms, 2020-11-29:
> @openoms can we push the torifying of services to 1.7 where we have the milestone to go Tor by default. I want to build the final v1.6.2 release today?

yes absolutely. Closing this now and will follow up in another issue.

closed 2020-11-29 by openoms