raspiblitz/raspiblitz #1799: RC1 BTRFS error reports

rootzoll, 2020-11-23:
As @jodobear mentioned on https://github.com/rootzoll/raspiblitz/issues/1785#issuecomment-732292845 there were some problems on updating from a v1.5 BTRFS RaspiBlitz. 

So you removed the pen drive for RAID and had to insert a fresh sd card with v1.6.2rc1 because you were getting a `Ctrl+D or root password screen`. And then it was not automatically going into recovery/update but promted you to setup a new Blitz with the option of MIGRATION. Is that correct so far?

jodobear, 2020-11-24:
1. Generated lnd rescue file & MIGRATION file `>` Removed pen drive `>` truned off RAID & other suggestions per [#1683(comment)](https://github.com/rootzoll/raspiblitz/issues/1683#issuecomment-716034494) `>` Got `Ctrl+D or root password screen`.

2. flashed 1.6.2rc1 `>` 1.6.2rc1 on boot gives error: `(standard_in) 1: syntax error` `>` Attempted MIGRATION failed checksum `>` Attempted Recovery which resulted in errors as mentioned in [#1785(comment)](https://github.com/rootzoll/raspiblitz/issues/1785#issuecomment-732292845)

rootzoll, 2020-11-25:
hmmm this seems multiple errors combined. Is it possible for you to check the md5 hash sum of the downloaded migration file? It should be se same as in the filename. example: if the name is `raspiblitz-test-20-11-25-17-25-3b44423de8b46dcb94dc9de51c4acd07.tar.gz` then the md5 sum should be is `3b44423de8b46dcb94dc9de51c4acd07` otherwise the migration file was not downloaded correctly. The new v1.6.2 will advise the user to check md5 checksum and/or bytessize to prevent faulty downloads.

jodobear, 2020-11-25:
I did check the hash, and they are different. This was surprising for me as i don't understand how can they be different looking at the `migration` script. I still have the migration file on my blitz ssd, i redownloaded it and still they are different.

The user check should add more robustness to this.

I think i should resetup my blitz with the lnd rescue file. I doubt this setup is rescueable. Any comments?

rootzoll, 2020-11-25:
If the hash is different that should point into a problem during download/scp of the file. If the original rescue file on the TEMP partition of the HDD is also not matching its md5 filename part ... thats strange and maybe something went wrong on doing the md5 of the file before renaming it, but the part of the code doing this is looking good to me: https://github.com/rootzoll/raspiblitz/blob/b21a36e3b87320cd693899558fbb5c8dd371ac51/home.admin/config.scripts/blitz.migration.sh#L103 Anybody can see a problem here?

At this point I would advise a resetup. RC2 will be out today or tomorrow - it would be great of you can test if you are able to reproduce this problem and let me know.

jodobear, 2020-11-25:
Yep, the hashing script looks perfectly alright. I did not check the hash while the file was in `temp`. Will check it there and report. Also will test a fresh rc2 setup over the weekend.

Thanks again!

rootzoll, 2020-11-26:
v1.6.2RC2 is available - see: https://github.com/rootzoll/raspiblitz/issues/1785#issuecomment-734025137

jodobear, 2020-12-03:
I'm getting exactly the same error with 1.6.2

rootzoll, 2020-12-10:
Closing this issue - because its continued in #1849 fo next release.

closed 2020-12-10 by rootzoll