raspiblitz/raspiblitz #1800: RC1 Update error from v1.6.0 report

rootzoll, 2020-11-23:
As @The34Goon reported on https://github.com/rootzoll/raspiblitz/issues/1785#issuecomment-730441002 there was a problem in updating from a v1.6.0 RaspiBlitz

You describe that the system is not starting because of an unvalid `tor.active` in the `lnd.conf`. Can you remember ever done changes manually to the lnd.conf in the past on that RaspiBlitz? Because even old v1.5 script were not putting that option there - Tor was always activated as a starting parameter from the lnd systemd service. Just wondering how that option got into the lnd.conf.

openoms, 2020-11-23:
Has been solved, the issue was an incompatible USB -disk adapter.
 Copying from Telegram:
```
It's most likely the USB adapter that caused all this.

Tryin to set it up from scratch I get many input/output errors 
and the blockchain transfer stops with read-only filesystem after about 5minutes.

I had the usb-adapter-firmware fixed before(JMS578 firmware updater) and somehow messed it up with the update.
Got my friend to send me the UGREEN adapter.

...

I was using this:
https://i-tec.pro/en/produkt/mysafeu313-2/
```

rootzoll, 2020-11-23:
Thanks - closing issue.

closed 2020-11-23 by rootzoll