raspiblitz/raspiblitz #1801: Check to include electrs fix/update into 1.6.2

rootzoll, 2020-11-24:
https://github.com/romanz/electrs/pull/334

openoms, 2020-11-24:
The good thing is prometheus is notnused for electrs.

The bad is that the latest Electrum v4.0.5 reportedly does not connect to electrs. Did not test it yet.

Kixunil, 2020-11-24:
Electrum definitely refuses to connect to `electrs` older than https://github.com/romanz/electrs/commit/c88a0dc331eb16163276becf98fcc020565d97eb I'm recompiling to see if it works with the newest version.

Kixunil, 2020-11-24:
Confirmed, it's fixed in master. See also https://github.com/romanz/electrs/issues/331

romanz, 2020-11-25:
A new release is ready: https://crates.io/crates/electrs/0.8.6
Many thanks for reporting and investigating this issue!

rootzoll, 2020-11-25:
OK update for RaspiBlitz is merged and worked on testing. Now testing recovery.

rootzoll, 2020-11-25:
OK recovery worked - update will be part of v1.6.2 release. Closing issue.

closed 2020-11-25 by rootzoll