raspiblitz/raspiblitz #1805: SSD recommendations

openoms, 2020-11-25:
Aim to collect here SSD types which work reliable with the RPI4 powered only through the USB-C port.

See this comment to start:
https://github.com/rootzoll/raspiblitz/issues/905#issuecomment-581526545

More suggestions in this thread:
https://twitter.com/floristotle/status/1331715583939907585
and from this article:
https://www.anandtech.com/show/15887/the-samsung-870-qvo-1tb-4tb-ssd-review-qlc-refreshed/5

Please add if you have tried any specific models. 1TB and above preferred.

GoodKey, 2020-11-25:
RPi4 officially supports 1.1A for its USB ports (combined?).
So 5V x 1.1A means the SSD can have a peak power draw of 5,5W, theoretically. 
Unsure if the amperage on the back of SSD's is for 5V or for 3,3V.... 
I'll update this when I know more.

NilByte, 2020-11-26:
I'm running the Samsung 1TB SSD 870 QVO SATA III 2.5 inch (MZ-77Q1T0BW). Picked it for the decent transfer rates and the low power consumption. 

https://villman.com/Product-Detail/samsung_1tb-ssd-870-qvo-sata-iii-25-inch-mz-77q1t0bw

>Environment
Average Power Consumption (system level) Average: 2.2 W *Maximum: 4.0 W (Burst mode)* Actual power consumption may vary depending on system hardware & configuration
Power consumption (Idle) Max. 30 mW * Actual power consumption may vary depending on system hardware & configuration
Allowable Voltage 5V ± 5% Allowable voltage
Reliability (MTBF) 1.5 Million Hours Reliability (MTBF)
Operating Temperature 0 - 70 ? Operating Temperature
Shock 1,500 G & 0.5 ms (Half sine)>

Raspberry Pi 4 Model B with recommended touchscreen, this SSD and noting else connected is pulling between 5 and 8W on the wall depending on load.

SSD connected to ORICO 2.5 Inch USB C External Hard Drive Enclosure, USB 3.1 Gen 2 to SATA III 10 Gb/s and I have not had any trouble with it.

https://www.amazon.de/-/en/gp/product/B07NYC6LKB/ref=ppx_yo_dt_b_asin_title_o01_s01?ie=UTF8&psc=1

Ben10Brasileiro, 2021-03-02:
Deviece: Sabrent 256GB Rocket NVMe PCIe M.2 2280 Internal SSD High Performance Solid State Drive (SB-ROCKET-256) (https://www.amazon.com/Sabrent-Rocket-Internal-Performance-SB-ROCKET-1TB/dp/B07KGLN3HN/ref=sr_1_4?dchild=1&keywords=nvme%2Bsabrent%2Brocket&qid=1614375903&sr=8-4&th=1)
Enclosure: FIDECO M.2 NVME ＆ SATA SSD Enclosure Adapter https://www.amazon.com/FIDECO-NVME-SATA-Enclosure-Adapter/dp/B088M3TZ5Y/ref=sr_1_8?crid=2PBJ5073V65X3&dchild=1&keywords=fideco+m.2+nvme+external+ssd+enclosure&qid=1614642393&sprefix=fideco%2Caps%2C266&sr=8-8

Case 35 dollars + NVME 45 dollars (256GB) = 80 dollars.
The same nvme for 1TB costs 130 USD, 30 more than the sata, but the performance is another thing. Also, larger drives tend to have better speed, so my 256GB is slower than the 1TB one. 

As for the nvme, there is no good brand for 128GB, I suggest starting from 256GB and beyond. If running the Blitz, of course 1TB.
Why this enclosure: First, the chip Realtek RTL9210B, saw majorly excellent reviews about it. Second: the case have good dissipation and comes with thermal paste. Third: It is USB 3.2 Gen 2 10Gbps (not yet supported on the Pi). Fourth is UASP support. Fifth is no cable.
The negative: You will lose a USB 3.0 and a USB 2.0 port, only remaining a USB 2.0, for either the backup usb or a keyboard.
I tried the thiniest enclosure posible with good dissipation. A USB HUB for the pendrive for static channel and keyboard would be the solution, but wouldnt use it for a high storage device.

I used raspberry pi os with desktop cause I wanted to run their sd card test, and because could not fit the nvme and the sata to usb adapter on the two USB 3.0 ports at the same time. This does not invalidate the use of NVME for the Blitz, but using as a USB stick can loose some ports. Although, the most comman way is by using a enclosure with cable.
Jeff Geerling used this one of 24 USD - TDBT M.2 NVMe SSD Enclosure with Heat Sink, 10Gbps USB-C to PCIe NVMe M.2 Hard Drive Enclosure with Thermal Cooling Pad, NVMe M.2 Drive to USB-C External Storage Enclosure, Fits M-Key B+M Key NVMe SSD  https://www.amazon.com/TDBT-Enclosure-Thermal-Cooling-External/dp/B07TJT6W8K/ref=as_li_ss_tl?dchild=1&keywords=tdbt+nvme&qid=1596741190&s=electronics&sr=1-1&linkCode=ll1&tag=mmjjg-20&linkId=6cec59da75a3dc3f4a35f3bc73e6284b&language=en_US. His benchmarks here https://www.jeffgeerling.com/blog/2020/fastest-usb-storage-options-raspberry-pi

The best drive I saw on other benchmarks is this one (200 USD for 1TB, the most expensive but the best results) SAMSUNG 980 PRO 1TB PCIe NVMe Gen4 Internal Gaming SSD M.2 (MZ-V8P1T0B)  https://www.amazon.com/SAMSUNG-500GB-Internal-Gaming-MZ-V8P500B/dp/B08GLX7TNT/ref=sr_1_1?crid=1QL129MB83MKZ&dchild=1&keywords=samsung%2Bevo%2B980&qid=1614652692&sprefix=SAMSUNG%2BEVO%2B9%2Caps%2C300&sr=8-1&th=1

SATA problems (could be that is a 1TB device, vs 256GB nvme) - After some reading I noticed that nvme requires much less power than sata - Topic 3 of the article https://www.digitalcitizen.life/reasons-buy-nvme-ssd/
I noticed the fan make very high sounds when booting with sata, but not with nvme
The hdmi screen didnt appeared when booting.
Couldnt SSH, as raspberry pi os doesnt come with ssh enabled.
The only way I could see the hdmi screen was by connecting the drive after boot

Used this article to get the commands to benchmark: https://linuxconfig.org/how-to-benchmark-disk-performance-on-linux

# NVME through USB on the PI

### Raspberry pi diagnostics sd card speed test

Sabrent Gen3X4 with Fideco USB 3.2 Gen2x2 enclosure with Realtek RTL9210B board on the Raspberry Pi USB 3.0 port with 

prepare-file;0;0;277694;542
seq-write;0;0;276523;540
rand-4k-write;0;0;49573;12393
rand-4k-read;36088;9822;0;0
Sequential write speed 276523 KB/sec (target 100000) - PASS
Random write speed 12393 IOPS (target 500) - PASS
Random read speed 9022 IOPS (target 1500) - PASS
Test PASS

### Read speeds:

pi@raspberrypi:~ $ sudo hdparm -Ttv /dev/sda1
/dev/sda1:
 multcount     =  0 (off)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 31130/255/63, sectors = 524288, start = 8192
 Timing cached reads:   1544 MB in  2.00 seconds = 772.47 MB/sec
 Timing buffered disk reads: 256 MB in  0.74 seconds = 345.69 MB/sec

pi@raspberrypi:~ $ sudo hdparm -Ttv /dev/sda1
/dev/sda1:
 multcount     =  0 (off)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 31130/255/63, sectors = 524288, start = 8192
 Timing cached reads:   1544 MB in  2.00 seconds = 772.43 MB/sec
 Timing buffered disk reads: 256 MB in  0.75 seconds = 339.29 MB/sec

pi@raspberrypi:~ $ sudo hdparm -Ttv /dev/sda1
/dev/sda1:
 multcount     =  0 (off)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 31130/255/63, sectors = 524288, start = 8192
 Timing cached reads:   1552 MB in  2.00 seconds = 775.99 MB/sec
 Timing buffered disk reads: 256 MB in  0.74 seconds = 343.72 MB/sec

### Write speeds:

[... 171 lines cut ...]
 Timing buffered disk reads: 1322 MB in  3.00 seconds = 440.41 MB/sec

floor@debian:~$ sudo hdparm -tv /dev/sdb1
/dev/sdb1:
 multcount     =  0 (off)
 readonly      =  0 (off)
 readahead     = 256 (on)
 geometry      = 31130/255/63, sectors = 500115456, start = 2048
 Timing buffered disk reads: 1324 MB in  3.00 seconds = 440.76 MB/sec

root@debian:/tmp# dd if=/dev/zero of=/tmp/mnt/temp oflag=direct bs=128k count=1G
dd: error writing '/tmp/mnt/temp': No space left on device
1877829+0 records in
1877828+0 records out
246130671616 bytes (246 GB, 229 GiB) copied, 755.796 s, 326 MB/s

root@debian:/tmp# dd if=/dev/zero of=/tmp/mnt/temp oflag=direct bs=128k count=16k
16384+0 records in
16384+0 records out
2147483648 bytes (2.1 GB, 2.0 GiB) copied, 6.55825 s, 327 MB/s

root@debian:/tmp# dd if=/dev/zero of=/tmp/mnt/temp oflag=direct bs=128k count=32k
32768+0 records in
32768+0 records out
4294967296 bytes (4.3 GB, 4.0 GiB) copied, 12.9521 s, 332 MB/s

raspito, 2021-03-14:
hi guys, i am using the mx500 drive paired with your suggested enclosure with umbrel without any issue but really want to migrate to raspiblitz for more control - but it really seems as though there is some compatibility issue here. I'd really prefer not to buy another drive - has there been any progress made on understanding why there is an issue with the mx500 1TB drive?

openoms, 2021-03-14:
@raspito I had occassional freezes with the MX500 when powered only through the USBC3 port of the RPi with a 3A power supply.
Since I am running it powered separately with an X825 board and barrel connector it is working fine.
Since then there were firmware updates to the RPi so might have improved.
If you did not experience issues running other software imthe hardware ahould be fine, but if you experience something similar will know wht the issue is.

raspito, 2021-03-14:
thanks @openoms. how can i bump the firmware of the RPi? 

I can't fit the X825 board in my case, so ideally I'd like to get this thing running on the MX500 with stock hardware.

openoms, 2021-03-14:
@raspito if you are using the latest image of raspiblitz it is up-to-date.

xanoni, 2021-08-04:
I would add the Samsung T5 with 1TB or 2TB to this list. It's an USB3 drive that performs really well in benchmarks, it is typically on the 1st or 2nd page on PiBenchmarks:  https://pibenchmarks.com/fastest/1 or https://pibenchmarks.com/fastest/2

Besides, it is very compact and doesn't need a separate case / heatsink / cable.

Only one disclaimer, it needs an UDEV rule added for `fstrim` to work properly:

```bash
# idProduct may differ depending on storage size, see `dmesg`
sudo echo 'ACTION=="add|change", ATTRS{idVendor}=="04e8", ATTRS{idProduct}=="61f5", SUBSYSTEM=="scsi_disk", ATTR{provisioning_mode}="unmap"' >> /etc/udev/rules.d/10-uas-discard.rules
```
Results in:
```bash
# /etc/udev/rules.d/10-uas-discard.rules
ACTION=="add|change", ATTRS{idVendor}=="04e8", ATTRS{idProduct}=="61f5", SUBSYSTEM=="scsi_disk", ATTR{provisioning_mode}="unmap"
```

Alternatively, the following command should work for a non-persistent solution:
```bash
# please replace "sda" with your drive name
sudo echo unmap >  /sys/block/sda/device/scsi_disk/*/provisioning_mode
```

david-bakin, 2021-09-17:
BTW, T5 1Tb works for me just fine, newer model T7 1Tb does _not_ - the T7 is extremely flakey on the Pi, couldn't even copy the blockchain on to it.  I think it uses slightly more power, or maybe just demands slightly higher voltage - just enough that powering it through the RPi4's USB ports wasn't sufficient.  (Didn't try it though a powered hub.)

david-bakin, 2021-09-17:
@xanoni - your script for adding the trim rule doesn't work for me: `udevadm test` says "invalid inotify descriptor".  (idProduct and idVendor are correct).

I notice the `SUBSYSTEM` I get via `udevadm info -q all -a /dev/sda1` is `scsi` not `scsi_disk` but aside from that I don't know what I'm doing.  I don't even know if I'm running the correct `udevadm` command to test it:

```
udevadm test --action="change" /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.0/host0/target0:0:0/0:0:0:0/block/sda/sda1
```

xanoni, 2021-09-17:
@david-bakin sorry looking at this on my phone and might have missed something but just to make sure, do you have a Samsung T5 drive?

david-bakin, 2021-09-17:
The T5, and of course I confirmed with `dmesg` that the product&vendor ids are the same.  (This is from 1.7.0.)

```
=============================================================== T5 START
[    1.984600] usb 2-2: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[    2.014201] usb 2-2: New USB device found, idVendor=04e8, idProduct=61f5, bcdDevice= 1.00
[    2.014222] usb 2-2: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[    2.014234] usb 2-2: Product: Portable SSD T5
[    2.014246] usb 2-2: Manufacturer: Samsung
[    2.014258] usb 2-2: SerialNumber: 123456871A77
[    2.023243] systemd[1]: System time before build time, advancing clock.
[    2.027467] scsi host0: uas
[    2.028848] scsi 0:0:0:0: Direct-Access     Samsung  Portable SSD T5  0    PQ: 0 ANSI: 6
[    2.030920] sd 0:0:0:0: [sda] 1953525168 512-byte logical blocks: (1.00 TB/932 GiB)
[    2.031111] sd 0:0:0:0: [sda] Write Protect is off
[    2.031128] sd 0:0:0:0: [sda] Mode Sense: 43 00 00 00
[    2.031519] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    2.032207] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes
[    2.055729]  sda: sda1
[    2.058612] sd 0:0:0:0: [sda] Attached SCSI disk
=============================================================== T5 END
```

(Look at that serial number though! Did I get lucky `123456...` or is that just Samsung's idea of a joke?)

xanoni, 2021-09-18:
@david-bakin the serial number is Samsung's idea of a joke, all my T5 drives have the format `1234568XXXXX`.

I noticed that your drive uses `uas`, whereas mine say `UAS is ignored for this device, using usb-storage instead`. I'm not 100% sure if that's a good or a bad thing. I just know that it's faster than the CPU can do full disk encryption, and that it works with `fstrim`. Thus, I'm happy.

Did you set USB quirks in `/boot/cmdline.txt`? I've prepended the following to the line:
```
usb-storage.quirks=04e8:61f5:u
``` 

Thus, my dmesg says:
```
usb-storage 2-2:1.0: Quirks match for vid 04e8 pid 61f5: 800000
```

Maybe it's related to that.

david-bakin, 2021-09-18:
@xanoni - no, actually it was me not following your directions, it should have been `scsi_disk` not `scsi` after all.

Now `lsblk --discard /dev/sda` shows:
```
NAME   DISC-ALN DISC-GRAN DISC-MAX DISC-ZERO
sda           0      512B       4G         0
└─sda1        0      512B       4G         0
```
as it should (non-zero amounts in `DISC-GRAN` and `DISC-MAX`) and `sudo fstrim --verbose /mnt/hdd` seems to do something ...

Thanks!

xanoni, 2021-09-18:
NICE!  `fstrim -av` all week long now.