raspiblitz/raspiblitz #1806: Error message on display: "cannot create temp file for here-document: No space left on device"

rene78, 2020-11-26:
I am getting the error message below on my display. The SSD is only 44% full. Or could it be the SD card? I had this in the past and a restart of the Raspiblitz helped. But it now came back after around 2 months.
[image]

rootzoll, 2020-11-26:
Hi ... this should be fixed updating the RaspiBlitz. There is already a v1.6.2RC2 out with a final release coming soon.
https://github.com/rootzoll/raspiblitz/issues/1785

Just power down the device and power it up with a fresh sd card image. It will go thru recovery and after that this should be fixed.

rene78, 2020-11-26:
Hi Christian, gonna do it ASAP. Thanks a lot. Gonna close the issue again.

closed 2020-11-26 by rene78