raspiblitz/raspiblitz #1807: /var/log to /mnt/hdd

6502CAPITAL, 2020-11-26:
Question and/or feature option; how about moving /var/log to /mnt/hdd by default?

I've seen that a lot has been optimized already for performance and/or less SD card writes. But still /var/log is being written a lot and even more when you have enabled some additional services (lnd, btcpayserver, etc.)

Did a quick scan for a few hours on the disk IO statistics (sudo iotop -aok). One bigger part for writing to the SD card for example is the "rsyslogd -n -iNONE [rs:main Q:Reg]" process.

rootzoll, 2020-11-26:
In general that should be a good idea to reduce the writing in sd card @frennkie? But what will happen if HDD is not mounting or is mounting late in systemd race ... and that link is "broken" or "not ready" - would this create side effects? Because we just had this with linking the ssh keys to HDD with resulted in ssh sometimes not starting correctly (race condition) and people were not able to ssh into the rspberrypi.

6502CAPITAL, 2020-11-26:
Yes, that can be problematic if this is not executed in a controlled manner (on semi low level resources/services).

Interesting approach to automate moving /var/log on the fly (after initial booting for example):
https://askubuntu.com/questions/346493/can-you-mount-new-partition-to-var-log-without-rebooting