raspiblitz/raspiblitz #1813: JoinMarket update to 0.8.0 with SegWit oderbook

openoms, 2020-11-27:
Will test quickly, but this should be straightforward to update and get into the v1.6.2. Installation did not change and I am running from the master since a while.

openoms, 2020-11-27:
From https://github.com/JoinMarket-Org/joinmarket-clientserver/blob/master/docs/release-notes/release-notes-0.8.0.md#make-libsecp256k1-tests-optional-in-installation-script:
```
Make libsecp256k1 tests optional in installation script

This option is enabled with the --disable-secp-check parameter of install.sh, and is useful for people on more constrained devices who want to speed up installation.

aa3cec4 Make libsecp256k1 tests optional
```

There could be around 10 minutes spared from the JoinMarket installation by enabling this option, but it gives me piece of mind that the code doing the signing is tested on the spot. 
In the JoininBox the tests are skipped for updates and test installs, but every node runs it once at least.
It is a tradeoff maybe slightly too much on the side of security, but 10 minutes is acceptable. What do you think @rootzoll?

rootzoll, 2020-11-27:
People going down the JoinMarket rabbit hole I think appriciate that extra 10 minutes.

rootzoll, 2020-11-27:
Is there anything still to add or can I merge the PR?

openoms, 2020-11-27:
I think it is ok, running well. The functionality is the same. People will need to migrate wallets to bech32, but that is individual and out of the scope of ours. 
The good thing is that the new users will start with a  bech32 wallet already this way.

rootzoll, 2020-11-27:
OK PR merged. Doing final testing on my machine.

rootzoll, 2020-11-27:
OK looks good on my test. Closing for v1.6.2 release.

closed 2020-11-27 by rootzoll