raspiblitz/raspiblitz #1816: Ver 1.6.2rc2: "Update Tor" breaks Tor (will not connect)

tweetytoo, 2020-11-28:
Warning: I am a noob without really any decent understanding of coding, so apologies.

I have a Raspi Blitz (Fulmo) that _was_ working with 1.6.2rc2 

I then (foolishly?) chose "update tor from the source code" and many many screens filled with updates (see tor_update.txt)

After that (and even after a reboot), I have this error:

```
*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2020-11-28 19:09:51 CET; 34s ago
  Process: 5383 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
  Process: 5385 ExecStart=/usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735 ${lndExtraParameter} (code=exited, status=1/FAILURE)
 Main PID: 5385 (code=exited, status=1/FAILURE)
```

...so it seems that I've lost some important configuration? Or maybe the "updated" source code is incompatible with raspian? 

In any case, I can't seem to roll back the tor install and I don't have the hardware (new mini sd card or even a mini reader) to start again from a clean image. 

So I am hoping to repair, configure or roll back Tor.

Any ideas?

Thanks in advance!

[tor_update.txt](https://github.com/rootzoll/raspiblitz/files/5611012/tor_update.txt)

rootzoll, 2020-11-29:
@openoms do you see anything here we need to fix for v1.6.2 release?

openoms, 2020-11-29:
We tested the Tor update method successfully before.

I have advised on Telegram already that for now there is no way around having an SDcard writer to be able to flash clean images as a fallback and update.

tweetytoo, 2020-11-29:
Thanks for looking at this issue.

I just logged in (via terminal) and see this:

```
945 RESTARTS DETECTED - LND might be in a error loop
Nov 29 10:53:46 zPi lnd[1593]: dial tcp 127.0.0.1:9050: connect: connection refused
Nov 29 10:54:46 zPi lnd[2614]: dial tcp 127.0.0.1:9050: connect: connection refused
err='Tor tcp connection refused'
More Error Detail:
err='Tor tcp connection refused'
```

In response to sudo journalctl -u lnd, I get: 

```
-- Logs begin at Sat 2020-11-28 19:04:15 CET, end at Sun 2020-11-29 10:56:56 CET. --
Nov 28 19:06:49 zPi systemd[1]: Starting LND Lightning Daemon...
Nov 28 19:06:49 zPi systemd[1]: Started LND Lightning Daemon.
Nov 28 19:06:50 zPi lnd[3068]: dial tcp 127.0.0.1:9050: connect: connection refused
Nov 28 19:06:50 zPi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 28 19:06:50 zPi systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 28 19:07:51 zPi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Nov 28 19:07:51 zPi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 1.
Nov 28 19:07:51 zPi systemd[1]: Stopped LND Lightning Daemon.
Nov 28 19:07:51 zPi systemd[1]: Starting LND Lightning Daemon...
Nov 28 19:07:51 zPi systemd[1]: Started LND Lightning Daemon.
Nov 28 19:07:51 zPi lnd[3914]: dial tcp 127.0.0.1:9050: connect: connection refused
Nov 28 19:07:51 zPi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 28 19:07:51 zPi systemd[1]: lnd.service: Failed with result 'exit-code'.

[skipping 7000 lines...]

Nov 29 10:55:47 zPi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Nov 29 10:55:47 zPi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 945.
Nov 29 10:55:47 zPi systemd[1]: Stopped LND Lightning Daemon.
Nov 29 10:55:47 zPi systemd[1]: Starting LND Lightning Daemon...
Nov 29 10:55:47 zPi systemd[1]: Started LND Lightning Daemon.
Nov 29 10:55:47 zPi lnd[4197]: dial tcp 127.0.0.1:9050: connect: connection refused
Nov 29 10:55:47 zPi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 29 10:55:47 zPi systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 29 10:56:47 zPi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Nov 29 10:56:47 zPi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 946.
Nov 29 10:56:47 zPi systemd[1]: Stopped LND Lightning Daemon.
Nov 29 10:56:47 zPi systemd[1]: Starting LND Lightning Daemon...
Nov 29 10:56:47 zPi systemd[1]: Started LND Lightning Daemon.
Nov 29 10:56:47 zPi lnd[5315]: dial tcp 127.0.0.1:9050: connect: connection refused
Nov 29 10:56:47 zPi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 29 10:56:47 zPi systemd[1]: lnd.service: Failed with result 'exit-code'.
```

In response to curl --socks5 localhost:9050 --socks5-hostname localhost:9050 -s https://check.torproject.org/ | cat | grep -m 1 Congratulations | xargs <== I removed the extra ")" at the end

... I get _a line skip_ [no output]

So, pretty sure tor is not running and LND is pinging itself to death...

tweetytoo, 2020-11-29:
So there's no way to fix Tor?

openoms, 2020-11-29:
Can you run the above to know if Tor is broken at all?

openoms, 2020-11-29:
@rootzoll to double check ran the Tor update on a node which was not a fresh install, but updated to v1.6.2RC2 scripts. Got noe errors with lnd or else.
```
Finished running lintian.
# Stopping the tor.service before updating
# Update ...
(Reading database ... 107829 files and directories currently installed.)
Preparing to unpack tor_0.4.4.5-1~d10.buster+1_armhf.deb ...
Unpacking tor (0.4.4.5-1~d10.buster+1) over (0.4.2.7-1~d10.buster+1) ...
Setting up tor (0.4.4.5-1~d10.buster+1) ...
Something or somebody made /var/log/tor disappear.
Creating one for you again.
Processing triggers for systemd (241-7~deb10u4+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
# Starting the tor.service 
# Installed Tor version 0.4.4.5.
Starting the main menu ...
```

tweetytoo, 2020-11-29:
> Can you run the above to know if Tor is broken at all?

What "above"?

tweetytoo, 2020-11-29:
> @rootzoll to double check ran the Tor update on a node which was not a fresh install, but updated to v1.6.2RC2 scripts. Got no errors with lnd or else.

So, I've got some other "mix" of config and apps that's broken Tor? Perhaps that means I cannot fix the problem by removing or reinstalling tor (assuming I tried that)?

I can see how a clean restart is better, but it would be helpful to know what is broken :(

openoms, 2020-11-29:
See just:
`curl --socks5 localhost:9050 --socks5-hostname localhost:9050 -s https://check.torproject.org`

can remove Tor with:
`sudo apt remove tor`
`sudo apt purge tor`
then
`sudo apt install tor`

Anyway the solution here which does not take a disproportionate amount of time and effort is to get an SDcard writer and flash a clean image. 

The PATCH menu is recommended only to advanced users.

tweetytoo, 2020-11-29:
`curl --socks5 localhost:9050 --socks5-hostname localhost:9050 -s https://check.torproject.org`

results in no output.

Removing tor, I get this: 

```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
(Reading database ... 107852 files and directories currently installed.)
Removing tor (0.4.4.6-1~d10.buster+1) ...
Processing triggers for man-db (2.8.5-2) ...
```

How can I set local settings? I think that's an issue with Tor (see the log in [my first comment](https://github.com/rootzoll/raspiblitz/issues/1816#issue-752736761)).

I think I need to fix that before I try to reinstall Tor.

openoms, 2020-11-29:
not sure, can try setting in 
`sudo raspi-config`

tweetytoo, 2020-11-29:
Update: I tried (via sudo raspi-config) to edit local settings but got rejected:

```
Generating locales (this might take a while)...
  en_GB.UTF-8... done
  en_US.UTF-8... done
Generation complete.
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "C"
    are supported and installed on your system
```.

tweetytoo, 2020-11-29:
The pi is very unhappy about locale :(

`admin@zPi:~ $ sudo apt purge tor
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following package was automatically installed and is no longer required:
  torsocks
Use 'sudo apt autoremove' to remove it.
The following packages will be REMOVED:
  tor*
0 upgraded, 0 newly installed, 1 to remove and 2 not upgraded.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] Y
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
(Reading database ... 107817 files and directories currently installed.)
Purging configuration files for tor (0.4.4.6-1~d10.buster+1) ...
Processing triggers for systemd (241-7~deb10u4+rpi1) ...`

tweetytoo, 2020-11-29:
Seems there's an error caused by OSX setting locale!

https://askubuntu.com/questions/599808/cannot-set-lc-ctype-to-default-locale-no-such-file-or-directory

tweetytoo, 2020-11-29:
I turned off OSX's "interference" with locale, installed Tor and rebooted.

With `curl --socks5 localhost:9050 --socks5-hostname localhost:9050 -s https://check.torproject.org`

...I got a bunch of html (including "Congratulations. This browser is configured to use Tor.")

Now my screen says: `Public null:8333 10 peers` for Tor. Not sure if that's working?

And LND isn't running ;(

```
*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2020-11-29 13:45:38 CET; 47s ago
  Process: 21642 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
  Process: 21648 ExecStart=/usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735 ${lndExtraParameter} (code=exited, status=1/FAILURE)
 Main PID: 21648 (code=exited, status=1/FAILURE)
```

tweetytoo, 2020-11-29:
Ok, so LND cannot connect to Tor. Suggestions on how to configure that?

```
*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sun 2020-11-29 13:58:45 CET; 45s ago
  Process: 15372 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
  Process: 15374 ExecStart=/usr/local/bin/lnd --tor.active --tor.streamisolation --tor.v3 --listen=127.0.0.1:9735 ${lndExtraParameter} (code=exited, status=1/FAILURE)
 Main PID: 15374 (code=exited, status=1/FAILURE)

Nov 29 13:58:45 zPi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 29 13:58:45 zPi systemd[1]: lnd.service: Failed with result 'exit-code'.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Sun 2020-11-29 13:49:33 CET, end at Sun 2020-11-29 13:59:31 CET. --
Nov 29 13:57:12 zPi systemd[1]: Started LND Lightning Daemon.
Nov 29 13:57:28 zPi lnd[13842]: unable to initialize tor controller: unable to connect to Tor server: dial tcp 127.0.0.1:9051: connect: connection refused
Nov 29 13:57:28 zPi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 29 13:57:28 zPi systemd[1]: lnd.service: Failed with result 'exit-code'.
Nov 29 13:58:28 zPi systemd[1]: lnd.service: Service RestartSec=1min expired, scheduling restart.
Nov 29 13:58:28 zPi systemd[1]: lnd.service: Scheduled restart job, restart counter is at 5.
Nov 29 13:58:28 zPi systemd[1]: Stopped LND Lightning Daemon.
Nov 29 13:58:29 zPi systemd[1]: Starting LND Lightning Daemon...
Nov 29 13:58:29 zPi systemd[1]: Started LND Lightning Daemon.
Nov 29 13:58:45 zPi lnd[15374]: unable to initialize tor controller: unable to connect to Tor server: dial tcp 127.0.0.1:9051: connect: connection refused
Nov 29 13:58:45 zPi systemd[1]: lnd.service: Main process exited, code=exited, status=1/FAILURE
Nov 29 13:58:45 zPi systemd[1]: lnd.service: Failed with result 'exit-code'.
```

adevoss, 2020-11-29:
On 1.6.1 fresh install on vm and running build_sdscard.sh
```
dial tcp 127.0.0.1:9050: connect: connection refused
err='Tor tcp connection refused'
```
Should I post here or create a new issue? Haven't tried things mentioned above yet.

rootzoll, 2020-11-29:
@adevoss please make a new issue. thx.

openoms, 2020-11-29:
@tweetytoo if you have updated with PATCH from 1.6.0 this wont work without deleting lines form the config files and reinstalling Tor. Changing the password_B from the menu might work to as it is synchronising the Tor password accross the services.

But honestly just get a 10EUR SDcard writer and flash a clean image, you will need it to update later anyway.

tweetytoo, 2020-11-29:
> @tweetytoo if you have updated with PATCH from 1.6.0 this wont work without deleting lines form the config files and reinstalling Tor. Changing the password_B from the menu might work to as it is synchronising the Tor password accross the services.

I will try the password change.
 
> But honestly just get a 10EUR SDcard writer and flash a clean image, you will need it to update later anyway.

Ordered. I've learned my lesson. I'm just trying to "find the bug" but I agree that it may be weird/waste of time. I still have a few days to bang my head on the wall ;)

Thanks for the help!

tweetytoo, 2020-12-03:
I ended up flashing and replacing the microSD card with a new version of 1.6.1. Everything is working again. 

Now I need to figure out how to update various packages without causing a conflict.

closed 2020-12-03 by tweetytoo

openoms, 2020-12-03:
Use the UPDATE menu, just not the PATCH.