raspiblitz/raspiblitz #1817: custom_install(s).sh file name (1.6.1)

adevoss, 2020-11-28:
When installing 1.6.1 I noticed the difference in names in the cp command.
It all works but a bit confusing. To me.
Maybe not renaming but just copy the file?

rootzoll, 2020-11-28:
Hmmm .. I think it was was always `custom-installs.sh` (not with an underline). See FAQ on it here:
https://github.com/rootzoll/raspiblitz/blob/v1.6/FAQ.md#why-do-i-need-to-re-burn-my-sd-card-for-an-update

adevoss, 2020-12-03:
My point is not hyphen/underline (you're right it is hyphen) but the renaming of the file.
_bootstrap.provision.sh
`line 549: sudo cp -av /mnt/hdd/app-data/custom-installs.sh /home/admin/custom-install.sh >> ${logFile}`

adevoss, 2020-12-03:
Edited the title from underscore to hyphen.

rootzoll, 2020-12-10:
> My point is not hyphen/underline (you're right it is hyphen) but the renaming of the file.
_bootstrap.provision.sh line 549: sudo cp -av /mnt/hdd/app-data/custom-installs.sh /home/admin/custom-install.sh >> ${logFile}

This is the script just making a work copy of that script to the admin home folder - it gets deleted in line 554 directly after excution. So main location of the script stays at `/mnt/hdd/app-data/custom-installs.sh`. Does that answere your question?

adevoss, 2020-12-10:
Sorry, no. I know it's not a bug, it's only a minor minor thing.
On the hdd the file is called 'custom-installs.sh' (watch the s before the .) and in the /home/admin directory it is 'custom-install.sh'.
Overlooked that it got deleted afterwards. That's making it even more minor.

rootzoll, 2020-12-10:
OK now I get your point ;) .. will mark this for v1.7 to change.

rootzoll, 2021-03-24:
OK change done ... make final testing if custom install script still works on recovery/update.

rootzoll, 2021-04-19:
OK final testing looks good .. will be part of v1.7 release.

closed 2021-04-19 by rootzoll