raspiblitz/raspiblitz #1818: [Feature Request] Add ION - Scalable Decentralised Identifiers on Bitcoin

rosskdavis, 2020-11-29:
This would be great addition and it appears they have a great point that it is most likely the missing link for Dapps

https://youtube.com/watch?v=kUb7ENx6Nyk   # a short overview

https://aka.ms/ion-beta-post    # Github etc.linked off this

rootzoll, 2020-11-29:
The idea to add it to the RaspiBlitz you can find it already as an issue here: #1038 - it would be great is someone treis this out. On how to contribute to RaspiBlitz see here: https://www.youtube.com/watch?v=ZVtZepV3OfM

rootzoll, 2021-03-25:
ION is now live: https://twitter.com/csuwildcat/status/1375125876736548870?s=20

ros-mac, 2021-03-26:
https://twitter.com/csuwildcat/status/1375467632602705923  # guide for running on Pi4

cstenglein, 2023-06-07:
Since this project is [archived since Jul 21, 2022](https://github.com/henkvancann/ion) I'm closing this for now.

closed 2023-06-07 by cstenglein