raspiblitz/raspiblitz #1821: [1.6.1] Tor connection refused

adevoss, 2020-11-29:
Fresh install on vm and running build_sdscard.sh
Read similar issue and purged and reinstalled tor (0.4.4.6-1~d10.buster+1)
```
dial tcp 127.0.0.1:9050: connect: connection refused
err='Tor tcp connection refused'
```
```
netstat -tulpn
tcp        0      0 127.0.0.1:9050          0.0.0.0:*               LISTEN      21460/tor
```

adevoss, 2020-11-29:
```
curl --socks5 localhost:9050 --socks5-hostname localhost:9050 -s https://check.torproject.org/ | cat | grep -m 1 Congratulations | xargs
Congratulations. This browser is configured to use Tor.
```

rootzoll, 2020-12-10:
There will be a big refactor of the setup process for v1.7 ... will make this a reminder to recheck the VM build on it.

openoms, 2021-09-12:
possibly related: https://github.com/rootzoll/raspiblitz/issues/2531

openoms, 2021-11-30:
Closing as should be fixed by afbc484. Continue in #2531 if the issue persists.

closed 2021-11-30 by openoms