raspiblitz/raspiblitz #1822: Dropbox did not do backup

AntonKlein, 2020-11-29:
The Dropbox did not do the backup. I did it exactly the way described here in Github.
I'm  wondering because after copying the token the StaticChannelBackup Dropbox is not selected with a  "*" (see picture below). 

Can anyone help me?  My mac is running with "HighSierra" 

[image]

openoms, 2020-11-29:
What version of the SDcard image you are using?
Will need to achieve that to have that SCB on Dropbox option selected. Waht happens if you select it and press OK?

AntonKlein, 2020-11-29:
> What version of the SDcard image you are using?
> Will need to achieve that to have that SCB on Dropbox option selected. Waht happens if you select it and press OK?

and here I pasted the token which was generated under "settings" in the Dropbox (and yes I copied all the letters :-) 
[image]

rootzoll, 2020-11-29:
Is that with a v1.6.2RC or with 1.6.1?

AntonKlein, 2020-11-29:
It’s with 1.6.1.

um 18:21:

rootzoll, 2020-11-30:
v1.6.2 which is close to release will have a fix for that:
https://github.com/rootzoll/raspiblitz/issues/1457

If you have a node to run tests on you can even give the release cadidate a test:
https://github.com/rootzoll/raspiblitz/issues/1785

I am closing this issue as a duplicate - please use #1457 for further report.

closed 2020-11-30 by rootzoll