raspiblitz/raspiblitz #1824: PyBlock errors on 1.6.1

JimBLogic, 2020-11-30:
The first time I installed PyBlock on my RaspiBlitz everything appeared to work fine, but then I tried the option to "Show confirmations from a transaction" with a valid Tx Id that I could verify was confirmed through RTL, on PyBlock I got the following error:

``` error code: -5
error message:
No such mempool transaction. Use -txindex or provide a block hash to enable blockchain transaction queries. Use gettransaction for wallet transactions.
Traceback (most recent call last):
  File "/home/pyblock/PyBLOCK/PyBlock.py", line 2748, in <module>
    menuSelection()
  File "/home/pyblock/PyBLOCK/PyBlock.py", line 1039, in menuSelection
    menu()
  File "/home/pyblock/PyBLOCK/PyBlock.py", line 318, in menu
    menuA(input("\033[1;32;40mSelect option: \033[0;37;40m"))
  File "/home/pyblock/PyBLOCK/PyBlock.py", line 2232, in menuA
    getrawtx()
  File "/home/pyblock/PyBLOCK/PyBlock.py", line 197, in getrawtx
    lsdb = lsda[-3]
IndexError: list index out of range ```

I then had a yellow message on the screen prompting me to select "U" to update PyBlock I did so, it asked me to log out and back in the program and I still couldn't use the "Show confirmations from a transaction" and this time instead of an error it would send me back to the main screen of PyBlock.

I decided to disable and re-enable Pyblock to check for installation bugs.
This is the text I got while re-enabling PyBlock:

``` *** INSTALL pyblocks***
Adding user `pyblock' ...
Adding new group `pyblock' (1004) ...
Adding new user `pyblock' (1004) with group `pyblock' ...
Creating home directory `/home/pyblock' ...
Copying files from `/etc/skel' ...
Cloning into '/home/pyblock/PyBLOCK'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 1641 (delta 3), reused 4 (delta 2), pack-reused 1632
Receiving objects: 100% (1641/1641), 1.30 MiB | 574.00 KiB/s, done.
Resolving deltas: 100% (902/902), done.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting art (from -r requirements.txt (line 5))
  Downloading https://files.pythonhosted.org/packages/a3/fa/eedb425a69ca6652b96423dcb73555a05c70d1f78ad039f2b856282c8300/art-5.0-py2.py3-none-any.whl (560kB)
    100% |████████████████████████████████| 563kB 744kB/s
Collecting qrcode (from -r requirements.txt (line 6))
  Downloading https://files.pythonhosted.org/packages/42/87/4a3a77e59ab7493d64da1f69bf1c2e899a4cf81e51b2baa855e8cc8115be/qrcode-6.1-py2.py3-none-any.whl
Requirement already satisfied: requests in /usr/lib/python3/dist-packages (from -r requirements.txt (line 7)) (2.21.0)
Requirement already satisfied: psutil in /usr/lib/python3/dist-packages (from -r requirements.txt (line 8)) (5.5.1)
Requirement already satisfied: simplejson in /usr/lib/python3/dist-packages (from -r requirements.txt (line 9)) (3.16.0)
Requirement already satisfied: certifi in /usr/lib/python3/dist-packages (from -r requirements.txt (line 10)) (2018.8.24)
Requirement already satisfied: chardet in /usr/lib/python3/dist-packages (from -r requirements.txt (line 11)) (3.0.4)
Requirement already satisfied: idna in /usr/lib/python3/dist-packages (from -r requirements.txt (line 12)) (2.6)
Collecting python-gnupg (from -r requirements.txt (line 13))
  Downloading https://files.pythonhosted.org/packages/cb/85/8a1588a04172e0853352ecfe214264c65a62ab35374d9ad9c569cf94c2a3/python_gnupg-0.4.6-py2.py3-none-any.whl
Collecting sseclient-py (from -r requirements.txt (line 15))
  Downloading https://files.pythonhosted.org/packages/6a/89/200c59842a7ce06c47d6cf2ab7598110c39909f5dc30c633b9a049acc262/sseclient_py-1.7-py2.py3-none-any.whl
Requirement already satisfied: urllib3 in /usr/lib/python3/dist-packages (from -r requirements.txt (line 16)) (1.24.1)
Collecting lnpay-py (from -r requirements.txt (line 17))
  Downloading https://www.piwheels.org/simple/lnpay-py/lnpay_py-0.1.0-py3-none-any.whl
Collecting xmltodict (from -r requirements.txt (line 18))
  Downloading https://files.pythonhosted.org/packages/28/fd/30d5c1d3ac29ce229f6bdc40bbc20b28f716e8b363140c26eff19122d8a5/xmltodict-0.12.0-py2.py3-none-any.whl
Collecting python-cfonts (from -r requirements.txt (line 19))
  Downloading https://files.pythonhosted.org/packages/93/cf/ef6f1d025965512d625408ff8a3999131be4726f6ad3aa3bdbd884d8c306/python_cfonts-1.4.0-py2.py3-none-any.whl
Collecting termcolor (from -r requirements.txt (line 20))
  Downloading https://www.piwheels.org/simple/termcolor/termcolor-1.1.0-py3-none-any.whl
Collecting pycoingecko (from -r requirements.txt (line 21))
  Downloading https://files.pythonhosted.org/packages/cf/b6/abf90df392a36fad22c86dfd5d089e2b9ea748a08ca54004113572cf51b0/pycoingecko-1.4.0-py3-none-any.whl
Collecting googleapis-common-protos==1.52.0 (from -r requirements.txt (line 22))
  Downloading https://files.pythonhosted.org/packages/03/74/3956721ea1eb4bcf7502a311fdaa60b85bd751de4e57d1943afe9b334141/googleapis_common_protos-1.52.0-py2.py3-none-any.whl (100kB)
    100% |████████████████████████████████| 102kB 3.1MB/s
Requirement already satisfied: grpcio in /usr/local/lib/python3.7/dist-packages (from -r requirements.txt (line 23)) (1.29.0)
Collecting protobuf==3.13.0 (from -r requirements.txt (line 24))
  Downloading https://files.pythonhosted.org/packages/3b/ba/276d817d4b0184e544a9295e14e95f6148af5c5652033d30eb63d2c13bfa/protobuf-3.13.0-py2.py3-none-any.whl (438kB)
    100% |████████████████████████████████| 440kB 934kB/s
Collecting six==1.15.0 (from -r requirements.txt (line 25))
  Downloading https://files.pythonhosted.org/packages/ee/ff/48bde5c0f013094d729fe4b0316ba2a24774b3ff1c52d924a8a4cb04078a/six-1.15.0-py2.py3-none-any.whl
Requirement already satisfied: colorama in /usr/lib/python3/dist-packages (from python-cfonts->-r requirements.txt (line 19)) (0.3.7)
Requirement already satisfied: setuptools in /usr/lib/python3/dist-packages (from protobuf==3.13.0->-r requirements.txt (line 24)) (40.8.0)
lnpay-py 0.1.0 has requirement requests>=2.22.0, but you'll have requests 2.21.0 which is incompatible.
Installing collected packages: art, six, qrcode, python-gnupg, sseclient-py, lnpay-py, xmltodict, python-cfonts, termcolor, pycoingecko, protobuf, googleapis-common-protos
  The script qr is installed in '/home/pyblock/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  The script cfonts is installed in '/home/pyblock/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed art-5.0 googleapis-common-protos-1.52.0 lnpay-py-0.1.0 protobuf-3.13.0 pycoingecko-1.4.0 python-cfonts-1.4.0 python-gnupg-0.4.6 qrcode-6.1 six-1.15.0 sseclient-py-1.7 termcolor-1.1.0 xmltodict-0.12.0
Reading package lists... Done
Building dependency tree
Reading state information... Done
hexyl is already the newest version (0.4.0-1).
The following package was automatically installed and is no longer required:
  libbluetooth3
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 134 not upgraded.
# Usage: https://github.com/curly60e/pyblock/blob/master/README.md
# To start type: 'sudo su pyblock' in the command line.
# Then pyblock
# To exit the user - type 'exit' and press ENTER ```

I also tested through PyBlock the Bitcoin-cli Console some commands and got these results:

``` console $>: getrawtransaction txid 92800a1fd9b05df3013f07085aa5a594daadb9b676f087a8e5d8994243319d3e
error: Error parsing JSON:92800a1fd9b05df3013f07085aa5a594daadb9b676f087a8e5d8994243319d3e

console $>: txid 92800a1fd9b05df3013f07085aa5a594daadb9b676f087a8e5d8994243319d3e
error code: -32601
error message:
Method not found

console $>: gettransaction 92800a1fd9b05df3013f07085aa5a594daadb9b676f087a8e5d8994243319d3e
error code: -32601
error message:
Method not found ```

My debuglog:  https://termbin.com/928o

openoms, 2020-11-30:
You have a disk corruption problem:
```
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - non-canonical ReadCompactSize(): iostream error at FlatFilePos(nFile=1355, nPos=21187681)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1355, nPos=59042769)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - non-canonical ReadCompactSize(): iostream error at FlatFilePos(nFile=1355, nPos=21187681)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1355, nPos=127594718)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1356, nPos=38850633)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1356, nPos=102005359)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Errors in block header at FlatFilePos(nFile=1356, nPos=90374866)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1356, nPos=111068526)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1356, nPos=85228219)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Errors in block header at FlatFilePos(nFile=1356, nPos=90374866)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1357, nPos=35131240)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1357, nPos=12589266)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Errors in block header at FlatFilePos(nFile=1357, nPos=131616458)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1356, nPos=126946765)
2020-11-30T13:34:51Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1357, nPos=103942162)
```
Proceed with backup and reset blockhain and change or format the disk.

nyxnor, 2021-04-18:
Please close this.

closed 2021-04-19 by rootzoll