raspiblitz/raspiblitz #1829: Losing connection to Electrs after a week of uptime.

ascisco, 2020-12-01:
Hardware: Pi4 (4GB)
Software:  Raspiblitz 1.6.1

Ive been running a node just fine for a few months but am now noticing a problem when trying to connect to electrs.

I'm usually able to connect to my electrs server but after a week or so it stops connecting. When I open my electrum wallet it continually tries to initialize a sync but it never succeeds.

If I restart my node it fixes the problem, but after another week goes by the same thing starts to happen.

rootzoll, 2020-12-10:
v1.6.2 is out that contains a electrs upadte - please try and report if you still have problems with this.

ascisco, 2020-12-19:
Seems to be working fine now. Thank you.

closed 2020-12-19 by ascisco