raspiblitz/raspiblitz #1832: Monitor Bitcoin Nodes Network Connectivity

rootzoll, 2020-12-01:
From time to time I get reports of nodes that just dont start syncing the blockchain or get stuck. The background process should check if bitcoind gets stuck for longer then 10 minues with zero peers even tho when internet is up and maybe inject some peer nodes by calling the Bitnodes API or from a static long list of long standing nodes with e.g `bitcoin-cli addnode "IP:PORT" "onetry"` or if running behind TOR and for 1 hour reset the hidden services address with `sudo rm /mnt/storage/bitcoin/onion_private_key` and restarting bitcoind.

openoms, 2020-12-01:
> for 1 hour reset the hidden service address with `sudo rm /mnt/storage/bitcoin/onion_private_key` and restart bitcoind.

should restart Tor too with 
`sudo systemctl restart tor`
to build new Tor circuits.

Reseting the bitcoind .onion address from time-to-time is beneficial for privacy reasons too: the broadcasted transactions won't be tied together. (the `torthistx` command helps there too)

rootzoll, 2021-04-05:
I will move this to the v1.8 because blockchain sync (that is not stopping) is part if the first user UX.

nyxnor, 2021-04-07:
On the UX, an button/option "Change Tor circuit", that will remove the priv key when the user wants. Also, change the priv key of all of the other services when requested too, not just Tor. 
Just an idea.