raspiblitz/raspiblitz #1834: nginx: command not found after updating to v1.6.2

openoms, 2020-12-02:
From https://t.me/raspiblitz/45313
[image]

>getting this when checking electrs from the menu, tried uninstall, re-install already
$ sudo nginx -t
sudo: nginx: command not found
perhaps nginx wasn't installed during the update process

Running:
`sudo apt install nginx`
and
`restart`
fixed the issue.

nyxnor, 2021-05-17:
Maybe check for important services being uninstalled with the background.sh script and then installing them again.

openoms, 2021-05-18:
this has been probably an uncomplete recovery, nginx  is not uninstalled during any setup step.
Also no further reports of this so closing.

closed 2021-05-18 by openoms