raspiblitz/raspiblitz #1835: Does LND auto force close?

inpharmaticist, 2020-12-02:
I have (had) a channel that provided me a good amount of inbound liquidity and I noticed today that it force closed, seemingly from my end. Is there a setting in LND that will automatically force close a channel for any reason? If so, is there a way to change it?

[image]

openoms, 2020-12-06:
There is no setting to stop this, this could have been a stuck HTLC and LND seeking settllement on the chain.  
See Balance of Satoshis `bos closed` for an automatic analyses.

Will close since there is nothing actionable in RaspiBiltz to mitigate this.

closed 2020-12-06 by openoms

inpharmaticist, 2020-12-06:
Thanks. What could cause this? Is there a resource I could learn more about stuck htlcs from?

openoms, 2020-12-06:
Don' t know a specific resource on the topic, but some useful links are:
https://lightningwiki.net/index.php/How_Lightning_Works
https://blog.bitmex.com/lightning-network-part-6-over-60000-non-cooperative-channel-closures/
and the rather technical Lightning Network specification: https://github.com/lightningnetwork/lightning-rfc/blob/master/00-introduction.md