raspiblitz/raspiblitz #1836: bitcoind didn't start after upgrade to v1.6.2

glowacki-dev, 2020-12-02:
The upgrade seemed to go fine, all packages got installed, I signed-in and got prompted to set ssh password, all required services were enabled and the system rebooted itself. After reboot I could login but the process stopped at https://github.com/rootzoll/raspiblitz/blob/v1.6.2/home.admin/70initLND.sh#L30
I tried rebooting a few times, but it didn't resolve the issue. Logs indicated that bitcoind service was unable to obtain lock on `/mnt/hdd/bitcoin` and won't start. There was a `/mnt/hdd/bitcoin/.lock` file that I had to manually remove to get it to start.
I'm not sure if the service didn't stop correctly before upgrade (I followed the regular update process from the menu, so it should be fine though) or if it crashed at some point during upgrade, but the result was that it got stuck in a locked state and won't start.
It's probably an edge case that won't happen often, but maybe it's worth adding some checks for lockfile when bitcoind service is not yet running?

rootzoll, 2020-12-10:
Yes thanks for the report - next release should add a check for lock files and remove them on updates.

rootzoll, 2020-12-24:
PR #1902 merged .. this should delete the blocking file on boot. Will be part of v1.6.3 release - closing issue.

closed 2020-12-24 by rootzoll