raspiblitz/raspiblitz #1839: Upgraded from version 1.6.1 to version 1.6.2 and now LND seems stuck after opening the wallet

Smiggel, 2020-12-04:
I upgraded this morning from a stable good running 1.6.1. node to version 1.6.2. After the upgrade it seems that my LND is suddenly stuck. The last item I see in my LND log is: LNWL: Opened wallet. That message is there for over half a hour now. And the display is saying "Prepare sync" the whole time.

My Bitcoin DB is 100% and running fine.

Is it normal that LND takes this long before it continues? I don't remember this taking so long before...

rootzoll, 2020-12-04:
"LNWL: Opened wallet" is a good message. Does you bitcoind has peers and is getting new blocks? What is `lncli getinfo` is telling you?

Smiggel, 2020-12-04:
I see this when running "lncli getinfo":

> [lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"

Smiggel, 2020-12-04:
Electrum is no longer running too. Is there something wrong when the Bitcoin conf?

Smiggel, 2020-12-04:
I saw this in the logs:

> /dev/fd/63: line 21: syntax error near unexpected token `('
/dev/fd/63: line 21: `err='lnd.conf: missing d.rpcuser (needs to be same as set in .conf)''
WARN: LND needs re-setup

Smiggel, 2020-12-05:
Ok, got it working. I flashed a new SD card with the same image as before. And now it all runs again. No issues. Perhaps the previous SD card was not ok anymore.

closed 2020-12-05 by Smiggel