raspiblitz/raspiblitz #1841: Spelling and grammar mistakes in 1.6.1 `Prepare for Update` screen

shawnyeager, 2020-12-04:
```
BEWARE: Your Lightning node is now stopped. It's safe to backup the data and
restore it on a fresh RaspiBlitz. But once this Lightning node gets started
again or rebooted its not adviced to restore the backup file anymore because
it cointains then outdated channel data & can lead to loss of channel funds.
```

Suggest that it reads:

```
BEWARE: Your Lightning node is now stopped. It's safe to backup the data and
restore it on a fresh RaspiBlitz. But once this Lightning node gets started
again or rebooted, it's not advised to restore the backup file because
it contains then outdated channel data and can lead to loss of channel funds.
```

senf42, 2020-12-04:
I'd suggest:

BEWARE: Your Lightning node is now stopped. It's safe to backup the data and
restore it on a fresh RaspiBlitz. But once this Lightning node gets started
again or rebooted, it's not advised to restore the backup file because
it would contain outdated channel data and can lead to loss of channel funds.

rootzoll, 2020-12-10:
Thanks - I changed the text for the v1.7 release. Closing this issue.

closed 2020-12-10 by rootzoll