raspiblitz/raspiblitz #1842: Update to v1.6.2 wiped out Specter and electrs fails to start

shawnyeager, 2020-12-04:
So, my Specter config is gone – devices and wallets.

And electrs isn't starting with the following:

```
Dec 04 21:57:57 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, stat
Dec 04 21:57:57 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
Dec 04 21:58:57 raspberrypi systemd[1]: electrs.service: Service RestartSec=1min expired, sched
Dec 04 21:58:57 raspberrypi systemd[1]: electrs.service: Scheduled restart job, restart counter
Dec 04 21:58:57 raspberrypi systemd[1]: Stopped Electrs.
Dec 04 21:58:57 raspberrypi systemd[1]: Started Electrs.
Dec 04 21:58:57 raspberrypi systemd[16900]: electrs.service: Failed to execute command: No such
Dec 04 21:58:57 raspberrypi systemd[16900]: electrs.service: Failed at step EXEC spawning /home
Dec 04 21:58:57 raspberrypi systemd[1]: electrs.service: Main process exited, code=exited, stat
Dec 04 21:58:57 raspberrypi systemd[1]: electrs.service: Failed with result 'exit-code'.
```

shawnyeager, 2020-12-04:
Using Fully Noded, I can still see the wallets (perhaps obvious, as they're using Bitcoin Core), but I know unsure if and how to resort within Specter. 

Any ideas, @stepansnigirev?

rootzoll, 2020-12-04:
About electrs please try an off/on under SERVICES again. About specter what wallets did you set .. @openoms can you take a second look?

openoms, 2020-12-04:
Looks like the recovery process did not finish running or the SDcard has failed. 
Would flash again an monitor the recovery log for errors.
Specter is storing the wallets on the disk inside the bitcoin directory.

shawnyeager, 2020-12-04:
Thanks, @rootzoll. Uninstalled and reinstalled electrs; that worked. 

Regarding wallets, not sure I follow the question. I had added one ColdCard and two wallets to it.

shawnyeager, 2020-12-04:
@openoms The wallets are there. As noted above, I can see them via Fully Noded. However, they're gone within Specter.

openoms, 2020-12-04:
Reinstall Specter too.

shawnyeager, 2020-12-04:
I used a brand new SD card for 1.6.2. I still have my card with 1.6.1. Could I retrieve the Specter config from there?

openoms, 2020-12-05:
Boot with 1.6.1 make a Specter backup from the Specter settings, restore in 1.6.2.(it is ok the LND channel states are on the disk and no major version jump between the two.)

shawnyeager, 2020-12-05:
Thanks, @openoms. I was able to boot into 1.6.1 and export my Specter backup. However, when then booting back into 1.6.2, LND is failing to start after repeated reboots.

```
****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

8 RESTARTS DETECTED - LND might be in a error loop
Dec 05 21:10:03 raspberrypi lnd[31981]: dial tcp 127.0.0.1:9050: connect: connection refused
Dec 05 21:11:03 raspberrypi lnd[3773]: dial tcp 127.0.0.1:9050: connect: connection refused
err='Tor tcp connection refused'
More Error Detail:
err='Tor tcp connection refused'

-> Use command 'repair' and then choose 'BACKUP-LND' to make a just in case backup.
-> Use command 'debug' for more log output you can use for getting support.
-> Use command 'menu' to open main menu.
-> Have you tried to turn it off and on again? Use command 'restart'

admin@raspberrypi:~ $
```

Full debug log here: https://termbin.com/0gzmz

What's my best course of action, please?

openoms, 2020-12-05:
Tor failed which can debugged or updated , but the easiest option is to flash the v1.6.2 image fresh.

rootzoll, 2020-12-15:
PR merged. Putting on final testing. @openoms what is the smallest "fuction test" to check that the persitent data between updates is working?

openoms, 2020-12-15:
if a device is added and a wallet is created it should be restored after reinstall and SDcard change also

rootzoll, 2021-01-13:
test looked good. closing issue for v1.6.3 release

closed 2021-01-13 by rootzoll