raspiblitz/raspiblitz #1849: 1.6.2 BTRFS Error Reports

jodobear, 2020-12-07:
As mentioned in https://github.com/rootzoll/raspiblitz/issues/1785#issuecomment-735395411 I'm getting the same error on 1.6.2, as posted below. Creating new issue as it relates to 1.6.2.

> UPDATE on 1.6.2 RC2 with fresh sd card image, fresh USB3 & unformatted ssd(never reached format options stage):
> 
> * On startup `data mount failed` on top line where blitz displays `Raspiblitz` and version.
> * On ssh i get `(standard_in) 1: syntax error` while loading.
> * On selecting to go ahead with re-setup. After Bitcoin > Hostname > Pass A > Pass B > Tor i get the following error:
> 
> ```shell
> HDD was already initialized/prepared
> Now needs to be mounted
> 
> # *** 40addHDD.sh ***
> (standard_in) 1: syntax error
> mount: /mnt/hdd: wrong fs type, bad option, bad superblock on /dev/sda1, missing codepage or helper program, or other error.
> 
> # --> Linking drives/directories
> # hddCandidate='sda'
> # hddPartitionCandidate='sda2'
> FAIL blitz.datadrive.sh link --> no data drive mounted
> ```

Is there a solution for this? It seems Raspiblitz can't be installed on the same hardware with these errors once arise on BTRFS

rootzoll, 2020-12-10:
> It seems Raspiblitz can't be installed on the same hardware

Have you formatted the HDD on your laptop once and then retried a a new setup?

jodobear, 2020-12-22:
Finally managed to install it. It was a formatting issue. Thanks!

closed 2020-12-22 by jodobear