raspiblitz/raspiblitz #1850: Error rapport

sjoerd933, 2020-12-07:
I got following RaspiBlitz 1.6.0 error.
Who can help me ?

THanks!

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Mon 2020-12-07 19:03:28 GMT, end at Mon 2020-12-07 19:13:36 GMT. --
-- No entries --
1607368396 ERROR: Error found in Logs
1607368415 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
tail: cannot open '/mnt/hdd/bitcoin/debug.log' for reading: No such file or directory

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: inactive (dead)

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Mon 2020-12-07 19:03:28 GMT, end at Mon 2020-12-07 19:13:36 GMT. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
tail: cannot open '/mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log' for reading: No such file or directory

*** LAST NGINX LOGS ***
sudo journalctl -u nginx -b --no-pager -n20
-- Logs begin at Mon 2020-12-07 19:03:28 GMT, end at Mon 2020-12-07 19:13:36 GMT. --
Dec 07 19:11:44 Sjoerd systemd[1]: Starting A high performance web server and a reverse proxy server...
Dec 07 19:11:51 Sjoerd nginx[643]: nginx: [emerg] a duplicate listen [::1]:3002 in /etc/nginx/sites-enabled/rtl_tor.conf:5
Dec 07 19:11:51 Sjoerd systemd[1]: nginx.service: Control process exited, code=exited, status=1/FAILURE
Dec 07 19:11:51 Sjoerd systemd[1]: nginx.service: Failed with result 'exit-code'.
Dec 07 19:11:51 Sjoerd systemd[1]: Failed to start A high performance web server and a reverse proxy server.
--> CHECK CONFIG: sudo nginx -t
nginx: the configuration file /etc/nginx/nginx.conf syntax is ok
nginx: configuration file /etc/nginx/nginx.conf test is successful

- TOUCHSCREEN is OFF by config

*** LAST 20 LOOP LOGS ***
sudo journalctl -u loopd -b --no-pager -n20
-- Logs begin at Mon 2020-12-07 19:03:28 GMT, end at Mon 2020-12-07 19:13:37 GMT. --
-- No entries --

*** LAST 20 RTL LOGS ***
sudo journalctl -u RTL -b --no-pager -n20
-- Logs begin at Mon 2020-12-07 19:03:28 GMT, end at Mon 2020-12-07 19:13:37 GMT. --
-- No entries --

*** LAST 20 ElectRS LOGS ***
sudo journalctl -u electrs -b --no-pager -n20
-- Logs begin at Mon 2020-12-07 19:03:28 GMT, end at Mon 2020-12-07 19:13:37 GMT. --
-- No entries --

*** ElectRS Status ***
##### STATUS ELECTRS SERVICE
configured=1
serviceInstalled=1
serviceRunning=0
tipSynced=0
initialSynced=0
electrumResponding=0
infoSync='Not running - check: sudo journalctl -u electrs'

- BTCPayServer is OFF by config
- LNbits is OFF by config
- Thunderhub is OFF by config
- SPECTER is OFF by config

*** MOUNTED DRIVES ***
Filesystem     Type     1K-blocks      Used Available Use% Mounted on
/dev/root      ext4      30450024   6180420  22982548  22% /
devtmpfs       devtmpfs   1966332         0   1966332   0% /dev
tmpfs          tmpfs      2000124         0   2000124   0% /dev/shm
tmpfs          tmpfs      2000124      8688   1991436   1% /run
tmpfs          tmpfs         5120         4      5116   1% /run/lock
tmpfs          tmpfs      2000124         0   2000124   0% /sys/fs/cgroup
tmpfs          tmpfs        32768         0     32768   0% /var/cache/raspiblitz
/dev/mmcblk0p1 vfat        258095     53177    204919  21% /boot
/dev/sda1      ext4     960378096 434048808 477474956  48% /mnt/hdd
tmpfs          tmpfs       400024         0    400024   0% /run/user/1000
tmpfs          tmpfs       400024         0    400024   0% /run/user/1001

*** WIFI Info ***
activated=0
connected=0
localip=''

*** HARDWARE TEST RESULTS ***
raspberryPi=4
powerFAIL=0
powerWARN=0
powerMIN='856300 microVolt'
tempFAIL=0
tempWARN=0
tempMAX='5300 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.178.122'
tempCelsius='50.6'
uptime=598
upsStatus='OFF'
startcountBlockchain=0
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=0
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

openoms, 2020-12-07:
make sure you use hardware form the recommended list or if you have an untested USB-disk adapter try it on the slow USB2 port (black).

nyxnor, 2021-04-18:
Please close this.

closed 2021-04-19 by rootzoll