raspiblitz/raspiblitz #1851: Cannot update or run joinmarket after installing new version

Zestforlife808, 2020-12-08:
After installing a new build mentioned by @openoms in https://github.com/rootzoll/raspiblitz/issues/1058#issuecomment-740245716 I can no longer run, install or update joinmarket.  Everything else works and I can finally use my pi again so that's progress.

If I perform a `sudo su joinmarket` it will launch me into the joininbox GUI but if I attempt to run any of the options It will fail.  For example, INFO returns 
```
Running the command:
 python wallet-tool.py   wallet.jmdat

Traceback (most recent call last):
  File "/home/joinmarket/joinmarket-clientserver/scripts/wallet-tool.py", line 2, in <module>
    from jmbase import jmprint
ModuleNotFoundError: No module named 'jmbase'

Fund the wallet on addresses labeled 'new' to avoid address reuse.

Press ENTER to return to the menu...

```

Attempting a full upgrade/reinstall from the joininbox UPDATE menu fails at the argon2_cffi portion of the install.  

```

Collecting argon2_cffi
  Using cached argon2-cffi-20.1.0.tar.gz (1.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /home/joinmarket/joinmarket-clientserver/jmvenv/bin/python3 /home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp3_zqqwcj
         cwd: /tmp/pip-install-_ttmqf2s/argon2-cffi_423dc42f34cc4f76a0af2ec9ade4008c
    Complete output (34 lines):
    Traceback (most recent call last):
      File "/home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
        main()
      File "/home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
        json_out['return_val'] = hook(**hook_input['kwargs'])
      File "/home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py", line 133, in prepare_metadata_for_build_wheel
        return hook(metadata_directory, config_settings)
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 140, in prepare_metadata_for_build_wheel
        self.run_setup()
      File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 126, in run_setup
        exec(compile(code, __file__, 'exec'), locals())
      File "setup.py", line 387, in <module>
        **keywords_with_side_effects(sys.argv)
      File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 145, in setup
        return distutils.core.setup(**attrs)
      File "/usr/lib/python3.7/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 447, in __init__
        k: v for k, v in attrs.items()
      File "/usr/lib/python3.7/distutils/dist.py", line 292, in __init__
        self.finalize_options()
      File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 739, in finalize_options
        ep.load()(self, ep.name, value)
      File "/tmp/pip-build-env-kdof5xum/overlay/lib/python3.7/site-packages/cffi/setuptools_ext.py", line 219, in cffi_modules
        add_cffi_module(dist, cffi_module)
      File "/tmp/pip-build-env-kdof5xum/overlay/lib/python3.7/site-packages/cffi/setuptools_ext.py", line 49, in add_cffi_module
        execfile(build_file_name, mod_vars)
      File "/tmp/pip-build-env-kdof5xum/overlay/lib/python3.7/site-packages/cffi/setuptools_ext.py", line 25, in execfile
        exec(code, glob, glob)
      File "src/argon2/_ffi_build.py", line 28, in <module>
        ffi = FFI()
      File "/tmp/pip-build-env-kdof5xum/overlay/lib/python3.7/site-packages/cffi/api.py", line 56, in __init__
        backend.__version__, backend.__file__))
    Exception: Version mismatch: this is the 'cffi' package version 1.14.4, located in '/tmp/pip-build-env-kdof5xum/overlay/lib/python3.7/site-packages/cffi/api.py'.  When we import the top-level '_cffi_backend' extension module, we get version 1.12.2, located in '/usr/lib/python3/dist-packages/_cffi_backend.cpython-37m-aarch64-linux-gnu.so'.  The two versions should be equal; check your installation.
    ----------------------------------------
ERROR: Command errored out with exit status 1: /home/joinmarket/joinmarket-clientserver/jmvenv/bin/python3 /home/joinmarket/joinmarket-clientserver/jmvenv/lib/python3.7/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp3_zqqwcj Check the logs for full command output.
Joinmarket was not installed. Exiting.
# installed JoinMarket v0.8.0
```

openoms, 2020-12-08:
did you try reinstalling joinmarket from raspiblitz SERVICES menu?

Zestforlife808, 2020-12-08:
Yes, sorry I did try that but I couldn't catch the log because the screen cache is reset when the script exits.  While watching it though, it says it installs the argon dependencies then the same block of red text comes up

openoms, 2020-12-09:
Can you try removing the conflicting cached file:
```
sudo rm /usr/lib/python3/dist-packages/_cffi_backend.cpython-37m-aarch64-linux-gnu.so
```
Both install options should work afterwards.

Zestforlife808, 2020-12-09:
Hmm that's odd.  I didn't have `_cffi_backend.cpython-37m-aarch64-linux-gnu.so` under dist-packages like the error was suggesting.  I ran the command anyway and then did a reinstall from the SERVICES menu and it worked like a charm!

I have all my funds back and my yield generator settings are intact.  I'm a happy camper.  Thank you for your help and quick response!
```
admin@raspberrypi:~ $ sudo rm /usr/lib/python3/dist-packages/_cffi_backend.cpython-37m-aarch64-linux-gnu.so
rm: cannot remove '/usr/lib/python3/dist-packages/_cffi_backend.cpython-37m-aarch64-linux-gnu.so': No such file or directory
admin@raspberrypi:~ $ sudo find / -name "*cffi*"
/var/lib/dpkg/info/python3-cffi-backend.list
/var/lib/dpkg/info/python3-cffi-backend.md5sums
/home/joinmarket/.local/lib/python3.7/site-packages/cffi-1.14.4.dist-info
/home/joinmarket/.local/lib/python3.7/site-packages/cffi
/home/joinmarket/.local/lib/python3.7/site-packages/cffi/_cffi_errors.h
/home/joinmarket/.local/lib/python3.7/site-packages/cffi/cffi_opcode.py
/home/joinmarket/.local/lib/python3.7/site-packages/cffi/__pycache__/cffi_opcode.cpython-37.pyc
/home/joinmarket/.local/lib/python3.7/site-packages/cffi/_cffi_include.h
/home/joinmarket/.local/lib/python3.7/site-packages/_cffi_backend.cpython-37m-aarch64-linux-gnu.so
/home/joinmarket/.cache/pip/wheels/0b/5a/01/5220481b9a4717ba183b738c9ee7d773ad32548555a339345a/cffi-1.14.4-cp37-cp37m-linux_aarch64.whl
/usr/lib/pypy/lib_pypy/_gdbm_cffi.pypy-41-aarch64-linux-gnu.so
/usr/lib/pypy/lib_pypy/cffi.egg-info
/usr/lib/pypy/lib_pypy/_syslog_cffi.pypy-41-aarch64-linux-gnu.so
/usr/lib/pypy/lib_pypy/_resource_cffi.pypy-41-aarch64-linux-gnu.so
/usr/lib/pypy/lib_pypy/_pwdgrp_cffi.pypy-41-aarch64-linux-gnu.so
/usr/lib/pypy/lib_pypy/_audioop_cffi.pypy-41-aarch64-linux-gnu.so
/usr/lib/pypy/lib_pypy/_sqlite3_cffi.pypy-41-aarch64-linux-gnu.so
/usr/lib/pypy/lib_pypy/__pycache__/_pypy_winbase_cffi.pypy-41.pyc
/usr/lib/pypy/lib_pypy/_pypy_winbase_cffi.py
/usr/lib/pypy/lib_pypy/cffi
/usr/lib/pypy/lib_pypy/cffi/_cffi_errors.h
/usr/lib/pypy/lib_pypy/cffi/cffi_opcode.py
/usr/lib/pypy/lib_pypy/cffi/__pycache__/cffi_opcode.pypy-41.pyc
/usr/lib/pypy/lib_pypy/cffi/_cffi_include.h
/usr/lib/pypy/lib_pypy/_curses_cffi.pypy-41-aarch64-linux-gnu.so
/usr/share/pypy/dist/pypy-cffi
/usr/share/doc/python3-cffi-backend
admin@raspberrypi:~ $ sudo find / -name "_cffi*"
/home/joinmarket/.local/lib/python3.7/site-packages/cffi/_cffi_errors.h
/home/joinmarket/.local/lib/python3.7/site-packages/cffi/_cffi_include.h
/home/joinmarket/.local/lib/python3.7/site-packages/_cffi_backend.cpython-37m-aarch64-linux-gnu.so
/usr/lib/pypy/lib_pypy/cffi/_cffi_errors.h
/usr/lib/pypy/lib_pypy/cffi/_cffi_include.h
find: ‘/proc/24893’: No such file or directory
find: ‘/proc/24915’: No such file or directory
```

closed 2020-12-09 by Zestforlife808