raspiblitz/raspiblitz #1852: RTL Loop error "HTTP request to an HTTPS server"

afcreamer, 2020-12-08:
Hello,

I've had trouble loading loop  and could not find other reports of this issue.

Raspiblitz v1.6.2
Loop 0.11.1-beta
RTL 0.9.3-beta

ERROR: Loop(327) => List Swaps Error: {"name":"StatusCodeError","statusCode":400,"message":"400 - \"Client sent an HTTP request to an HTTPS server.\\n\"","error":"Client sent an HTTP request to an HTTPS server.\n","options":{"url":"http://localhost:8081/v1/loop/swaps","rejectUnauthorized":

rootzoll, 2020-12-10:
Do you get the same error if you call RTL with `https://[LOCALIP]:3001` and `http://[LOCALIP]:3000`?

afcreamer, 2020-12-10:
Yes I just attempted it again on both and had the same result.

softsimon, 2020-12-11:
I have the same problem. Ride The Lightning can't connect to Loop after latest upgrade.

Trying manuall GET

```
admin@raspberrypi:~ $ curl http://localhost:8081/v1/loop/swaps
Client sent an HTTP request to an HTTPS server.
admin@raspberrypi:~ $ curl https://localhost:8081/v1/loop/swaps
curl: (60) SSL certificate problem: self signed certificate
More details here: https://curl.haxx.se/docs/sslcerts.html
```

openoms, 2020-12-11:
Just noting that you can always downgrade to the tested, working version by reinstalling from SERVICES.

Will need to investigate this with the RTL team I think, probably something with the API changed and might need to upgrade Loop too and adapt the config.

marssystems, 2020-12-11:
This was solved by the RTL team back in October.

https://github.com/Ride-The-Lightning/RTL/issues/515

openoms, 2020-12-16:
OK, PR incoming:
This fixes the HTTPS issue:
```
sudo cp /home/rtl/RTL/sample-RTL-Config.json /home/admin/RTL-Config.json
    sudo chown admin:admin /home/admin/RTL-Config.json
    sudo chmod 600 /home/admin/RTL-Config.json || exit 1
    PASSWORD_B=$(sudo cat /mnt/hdd/${network}/${network}.conf | grep rpcpassword | cut -c 13-)
    # modify sample-RTL-Config.json and save in RTL-Config.json
    sudo node > /home/admin/RTL-Config.json <<EOF
//Read data
var data = require('/home/rtl/RTL/sample-RTL-Config.json');
//Manipulate data
data.nodes[0].lnNode = '$hostname'
data.nodes[0].Authentication.macaroonPath = '/home/rtl/.lnd/data/chain/${network}/${chain}net/';
data.nodes[0].Authentication.configPath = '/home/rtl/.lnd/lnd.conf';
data.multiPass = '$PASSWORD_B';
data.nodes[0].Settings.userPersona = 'OPERATOR'
data.nodes[0].Settings.channelBackupPath = '/home/rtl/RTL-SCB-backup-$hostname'
data.nodes[0].Settings.swapServerUrl = 'https://localhost:8081'
//Output data
console.log(JSON.stringify(data, null, 2));
EOF
    sudo rm -f /home/rtl/RTL/RTL-Config.json
    sudo mv /home/admin/RTL-Config.json /home/rtl/RTL/
    sudo chown rtl:rtl /home/rtl/RTL/RTL-Config.json
 ```

openoms, 2020-12-16:
Now getting the error about the macaroon path, but the Loop macaroon authentication is not in RTL v0.9.3, so will need to wait for the next release, use RTL master or downgrade Loop to v0.9.0: https://github.com/lightninglabs/loop/releases

Macaroon auth is in:
https://github.com/Ride-The-Lightning/RTL/pull/543

rootzoll, 2020-12-21:
RTL 0.10.0 released TODO: Add to v1.6.3 release.
https://twitter.com/RTL_App/status/1340815355959267329?s=20

rootzoll, 2020-12-21:
PR merged. On final testing.

rootzoll, 2021-01-12:
testing looks good. closing for v1.6.3 release.

closed 2021-01-12 by rootzoll