raspiblitz/raspiblitz #1853: Add terminology definitions to Raspiblitz documentation

gked, 2020-12-08:
Could you please add terminology definitions for the screenshot below so that new comers understand the numbers displayed in 

Capacity - 
Commit Fee - 
Local - 
Remote - 
Base - 
PerMil - 

**There was also a proposal to add tooltips for these in RTL.**

[image]

rootzoll, 2020-12-10:
You can make that terminology definitions here and I can add them to the script.

gked, 2020-12-10:
@rootzoll I don't have all of the definitions. This was partially why I wanted to see them to be correctly defined in the app.
So, if someone else can vet what I have for correctness and completeness then I think, we could add them.
Anyways, this is what I have:

- Capacity - total sats in the channel (their side + your side)
- Commit Fee - the fee that's charged if either side of the channel closes (changes depending on fee market)
- Local - sats on your side of the channel (outbound liquidity)
- Remote - sats on their side of the channel (inbound liquidity)
- Base - not sure
- PerMil - routing fee I channel owner charges for allowing others to use his/her channel? Not sure.

rootzoll, 2020-12-14:
Base is the "Base-Fee" you charge for every forwarding on that channel.

gked, 2020-12-14:
> Base is the "Base-Fee" (millisatohis) you charge for every forwarding on that channel.

Thanks, I updated Base-Fee definition. PerMil is the only definition left,

rootzoll, 2020-12-15:
I updated for "PerMil" and wil add to the script output

rootzoll, 2020-12-15:
@gked changes comitted. You can test them with command MENU > PATCH > PATCH-MENU > BRANCH > set to `1853channelinfo` ... after that when you go back it should output the additional info.

gked, 2020-12-15:
@rootzoll just tested it and verified. Thanks!

rootzoll, 2020-12-15:
Merged. Closing for release.

closed 2020-12-15 by rootzoll