raspiblitz/raspiblitz #1855: SatStack: Connecting LedgerLive to your own FullNode

rootzoll, 2020-12-09:
https://twitter.com/P3b7_/status/1336349003714359297?s=20

henkisdabro, 2021-12-23:
Just checking in whether there are any updates to this issue? Is it currently possible to connect Ledger Live to the raspiblitz node? Upon trying, Ledger Live seems to find/connect via RPC, but I'm unsure how and where the Ledger satstack application should interact with raspiblitz, should it be installed on the raspiblitz itself?

adrienlacombe-ledger, 2022-09-16:
hi @rootzoll @henkisdabro, Adrien here from Ledger, I am working on satstack and would like to help if needed.
Currently I have only validated the following architecture: raspiblitz on a server (behind Tor or not) and satstack running on the same machine as Ledger Live. I guess it would also work if satstack runs on the same machine as raspiblitz **but** Tor would then need to be disabled, the trade off here would be that you open the satstack port to the Internet, which is not ideal IMHO.
Any questions, comments, let me know.
Thank you
Adrien

henkisdabro, 2022-09-24:
Hi there @adrienlacombe-ledger ! I just wanted to pop in to say I'm super excited that work is being done on this again, would love to see a good integration! Personally I don't have enough understanding of the inner working of raspiblitz itself, I'm more of a keen user and Ledger owner. Hence as such, hoping that the Raspiblitz team has some time over to provide you with the answers you need, and again thanks for writing in, I'll be following the thread!

rootzoll, 2022-09-24:
hi @adrienlacombe-ledger ..

> I guess it would also work if satstack runs on the same machine as raspiblitz but Tor would then need to be disabled, the trade off here would be that you open the satstack port to the Internet, which is not ideal IMHO.

For most services running on raspiblitz we open a local port (so people are able to connect directly in the same local network from their laptops) and on parralel setup a tor hidden service that provides the same service - so on clients that support tor the connection even works when the laptop is outside the local network - for example when people are traveling. Do you think this makes also sense for SatStack?

adrienlacombe-ledger, 2022-09-25:
thank you @rootzoll for the details! the local port completely makes sense yes! Unfortunately for the hidden services, I am still lacking a feature on Ledger Live to be able to connect it, but we are working on it ;)

rootzoll, 2022-09-26:
So for the first implementation (bounty) it will be OK to have this just for local IP.

adrienlacombe-ledger, 2022-09-30:
Excellent!

ziggie1984, 2022-10-18:
Hi I would like to tackle this on the weekend, so will work and report on Sunday

ziggie1984, 2022-10-18:
Looks like its still free, but nevertheless is already someone working on it ?

rootzoll, 2022-10-19:
@ziggie1984 its all yours for the weekend :) If you have any questions, happy to answer. And please report back - even if you dont see a fit after the weekend.

ziggie1984, 2022-10-25:
Ok just a short update:

was testing it successfully on the weekend, now I need to implement the bonus.script  and probably finish it this weekend!

ziggie1984, 2022-10-31:
@rootzoll should we build the sources on raspiblitz or download them from the github repo, building is very fast but most of the programs seem to be downloaded from release pages ?

rootzoll, 2022-11-02:
Normally all resources get downloaded from the github - best to fix to a commithash.

You should find examples of best practices if you take a look at the bonus template script:
https://github.com/rootzoll/raspiblitz/blob/v1.8/home.admin/config.scripts/bonus.template.sh

ziggie1984, 2022-11-04:
@adrienlacombe-ledger when using "importdescriptors" will bitcoin-core always rescan the wallet, it looks like it does so and it always starts from the beginning, did you experience something similar ?

ziggie1984, 2022-11-04:
Was testing satstack with only 1 descriptor and it rescans everything as soon as I restart, maybe bc I have zero funds on the descriptor will test now a descriptor on testnet where bitcoin-core will find founds

ziggie1984, 2022-11-06:
was able to fix the bug, by calling the `abortrescan` function before nuking processes in satstack. Now I can shutdown the lss and also restart it without bitcoind having any issues in loading the wallet

adrienlacombe-ledger, 2022-11-06:
@ziggie1984 could add this to satstack code somehow?

ziggie1984, 2022-11-06:
yes will do, I came also across a bitcoind bug, which I am reporting now, will open a PR in satstack to add this code!

adrienlacombe-ledger, 2022-11-06:
amazing! thank you so much!

ziggie1984, 2022-11-06:
https://github.com/bitcoin/bitcoin/issues/26463

ziggie1984, 2022-11-06:
@adrienlacombe-ledger can I reach you via twitter or telegram, I have some question related to ledger live and its usage of the satstack-api, or do you prefer to stay on github ?

adrienlacombe-ledger, 2022-11-06:
@ziggie1984 https://t.me/adrienlacombe

rootzoll, 2022-12-05:
@ziggie1984 were you able to clear your questions? Any updates?

ziggie1984, 2022-12-05:
Hi, yeah I have some updates, currently I am working on the satstack program to make the scanning of the blockchain more reliable, so integrating the current satstack program makes not sense imo, because it rescans the blockchain after every restart, I am about to change that in the satstack program but still a to do

ziggie1984, 2022-12-05:
but this isssue is on the radar, will give you the next steps during the week

ziggie1984, 2022-12-19:
ok first PR on satstack is out, next will follow, if this is merged (and the second one, still working on it) then integrating this in Blitz will be done for breakfast :)

https://github.com/LedgerHQ/satstack/pull/94

ziggie1984, 2022-12-22:
https://github.com/LedgerHQ/satstack/pull/95 when this final PR is merged, the integration into the blitz infrastructure will start

rootzoll, 2023-01-13:
Its merged - keep me updated :) I plan a next Release Candidate for RaspiBlitz v1.9 end of month ... do you think you might get a first prototype script ready by then to be included for people to test?

ziggie1984, 2023-01-13:
yes sure, this weekend on sunday integration should be finished, you have my word

rootzoll, 2023-01-18:
Was the weekend productive or do you still need some more days?

ziggie1984, 2023-01-19:
yes I need still some more days, had to test whats the best way to interact with the user, I decided that the user can either use the ledger-live setup and generate the config file needed for satstack there, but can also import the pubkeys with the provided script.

ziggie1984, 2023-01-21:
So while I was integrating all the stuff for the raspiblitz I saw that ledger-live itself only allows satstack to be running locally so there needs to be a change in the ledger-live software to support satstack running on the blitz, should be a small change for someone who is familiar with the ledger-live code base, could you connect us to one of the devs ? @adrienlacombe-ledger

ziggie1984, 2023-01-22:
Status-Update: So when connecting ledger-live to your own raspiblitz, I was not able to successfully verify the workflow on signet (receive and send a transaction connecting to my raspberrypi), somehow the ledger-live software fails syncing to the tip of the chain after publishing the transaction. So receiving and sending worked, but now the ledger-live software fails to sync. This is not a problem on the bridge side but on the ledger-live side. It needs investigation of the ledger-live software. So I am trying to dig into this, but not sure how fast I will dig into the ledger-live code. 

Also to make the user workflow way more comfortable some improvements need to be done in the ledger live software.

We need to decide how the user wants to get the xpubs on the raspiblitz. Do we want to write a script so that the user connects the device to the rpi, and the xpubs are dumped into a file needed for the bridge connection (tried this but had problems dumping the xpubs bc the device somehow was not found, guess I need some drivers here - so this way could be a problem for every user regarding which device they have etc.) or we let the user export them via the ledger-live software and copy the relevant file to the raspiblitz. This is a bit more user-involved but less error prone.

rootzoll, 2023-02-03:
@adrienlacombe-ledger so there seems more work on the ledger-live side to get this going - can you give un an idea how feasible it is to get this changes into ledger-live?

adrienlacombe-ledger, 2023-02-04:
@rootzoll could you please summarize what changes are required please? I would then take this to the appropriate team of devs. Thank you!

rootzoll, 2023-03-08:
@ziggie1984 can you get more specific what changes on the ledger-live software would be needed?

ziggie1984, 2023-03-22:
Thanks for the recommendation but no need for the bounty until this feature is in. Basically the main problem was an user interface problem, meaning that how to elegantly let the user start the satsback program and have everything needed without much interaction. I need some time in the next couple of days to anlayse what the missing parts were, will get back to you this week.