raspiblitz/raspiblitz #1856: JoinMarketQT GUI on the desktop: PySide2 Error

bc-vision, 2020-12-09:
Hi team,

I am trying to remote GUI (ssh tunnel) the JoinMarket instance on Raspiblitz with the JoinMarketQT GUI as described here: https://github.com/openoms/bitcoin-tutorials/blob/master/joinmarket/README.md ( in section "JoinMarketQT GUI on the desktop.")

Im getting always the same error: ( see screenshot as well)

(jmvenv) joinmarket@raspberrypi:~/joinmarket-clientserver/scripts $ python joinmarket-qt.py
Traceback (most recent call last):
  File "joinmarket-qt.py", line 27, in <module>
    from PySide2 import QtCore
ModuleNotFoundError: No module named 'PySide2'

I tried for testing purpose a standalone JoinMarket installation on fresh Ubuntu 20_04 and followed this video: https://www.youtube.com/watch?v=zTCC86IUzWo
Interesting outcome: First i got the exact same error on my ubuntu standalone as in raspiblitz ssh console,
but after second installation attempt it worked on ubuntu. 

I dont know how to fix it on rasperry, but i guess the solution is quite a simple one. Can anyone share experiance on that issue? 

My Environment: Rasperry4 8GB RAM, SSD 1TB, Raspiblitz 1.6.2 (latest) 

[image]

Cheers

openoms, 2020-12-09:
First of all try reinstalling JoinMarket rom.the RaspiBlitz SERVICES or JoininBox UPDATE.

bc-vision, 2020-12-10:
Thanks for the lightning fast reply :) 

JoinMarket reinstall over SERVICES menu unfortunately did not solve the problem. 

But i did another interesting observation and maybe it has to do with the issue.
My observation: Following this tutorial (https://www.youtube.com/watch?v=zTCC86IUzWo ) to install JoinMarket as standalone (just the testing use-case without raspiblitz) 

.... i figured out that the joinmarket.cfg file is not in the folder as described in video. 
The video shows that the .cfg file is in home/joinmarket-clientserver/scripts but in my installation (fresh Ubuntu 20.04 desktop and JoinMarket as standalone) it was in a hidden folder home/.joinmarket/joinmarket.cfg  

... maybe this is relevant i dont know, at least i share it here  maybe it helps. For sure it helps other people finding the joinmarket.cfg file on standalone installations following the video.

I will continue testing and keep you posted, in case somebody has an idea here or same problem, in best case a solution, please share. 

Cheers

bc-vision, 2020-12-10:
By the way, I don't know the perfect place to post this, but I did a nice build of RaspiBlitz with an mobile UPC.
For those who don't know UPC: https://en.wikipedia.org/wiki/Uninterruptible_power_supply
Here is a link to amazon: TinyURL with preview so its safe: https://preview.tinyurl.com/y3c6j583

This prevents your RaspiBlitz to get hard power offs what is always a critical topic.

Here is a pic of my build:
[image]

Technical stuff:
I know that this UPC has a USB power out specified with 2A. I tested the whole installation procedure of RaspiBlitz on a RaspberryPi 4, and observed the status with a current RaspiBlitz 1.6.2 installation was absolutely stable as it never got above 1,65 Ampere.  With USB backup stick and 2,5zoll SSD attached you are good to go. 

I had a USB Meter attached for those who want to test as well: 
https://preview.tinyurl.com/y2rphdbp

@openoms if you have a better place to share this info, please feel free to move or forward this post. Thanks. :)

Cheers

openoms, 2020-12-10:
Can't reproduce the issue:
[image]

Are you using the command from the QTGUI menu in JoininBox?
[image]

What OS?

If windows instead of the `qtgui` command you can try:
`/home/joinmarket/joinmarket-clientserver/jmvenv/bin/python /home/joinmarket/joinmarket-clientserver/scripts/joinmarket-qt.py`

bc-vision, 2020-12-12:
Ok thanks. After fresh SD card flash flashing it works perfect now. I cant reproduce it anymore it just works now. So all good. I am using currently windows for the QTGUI, so i have Xming installed. Thanks for link to the UPS topic and your support. 👍 

[image]

closed 2020-12-12 by bc-vision