raspiblitz/raspiblitz #1857: Seed phrase is left behind on filesystem if the setup is not run through completely

mmarsalko, 2020-12-10:
This could happen due to a loss of power or from ctrl-c'ing out during setup. Haven't tried reproducing this myself, but I'd noticed my seed phrase was in the file /home/admin/.seed.tmp.

I'd recommend putting important temporary files like this in /tmp so they disappear when power is lost or when the system reboots.

softsimon, 2020-12-10:
I too had the file `/home/admin/.seed.tmp` present containing my correct seed phrase.

Of what I know of I didn't do a incomplete setup. I did a regular step by step successful install of v1.6.1

tomstammis, 2020-12-10:
Looks to be the same for me. Seed is present, have not ctrl+c'ed out of the setup process, iirc.

rootzoll, 2020-12-10:
Thanks for the report. For v1.7 we will have a refactor of the setup process ... will add this issue as a reminder to move any setup related data to memory-based temp storage.

openoms, 2020-12-10:
just to clarify the file is securely erased if the script is not interrupted:
`ttps://github.com/rootzoll/raspiblitz/blob/64ee445b251ad3f59609e5f3cbcb24cd0e7ce664/home.admin/70initLND.sh#L202

gked, 2020-12-10:
+1 I was able to locate the file as well. Also went through the normal install process.
@openoms thanks for providing work around.

MiBy0815, 2020-12-11:
The issue description is kind of misleading, IMHO. Various people have confirmed this to happen without any interruption of the setup process

openoms, 2020-12-11:
@rootzoll this might worth a v1.6.3 build with no strong recommendation to update, but it would solve the issue for new users setting up the LND wallet the first time with the fresh image

rootzoll, 2020-12-11:
I haven opened a v1.6.3 release milestone and added this to be fixed.

rootzoll, 2020-12-15:
OK the branch 1857seeddelete contains a fix that makes sure the seed file gets deleted and moves it even to the RaspiBlitz RAM disk so that even after shredding it should leave no trace behind.

To be final tested and team discussed.

rootzoll, 2020-12-24:
OK this looks good on testing now - will be part of 1.6.3 - closing issue.

closed 2020-12-24 by rootzoll