raspiblitz/raspiblitz #1859: Blockchain does not sync after update to 1.6.2

JGalt040, 2020-12-10:
Hello,

I am a raspiblitz newbie and would really appreciate your help. After an update from 1.6.1 to 1.6.2 I do not get the blockchain to sync. After waiting for ~8 hours I still see this picture:
[image]

History:
1. installed raspiblitz with 1.6.1 - everything worked fined (without having really used it, but everything seemed to work)
2. tried upgrade to 1.6.2 and couldn't connect to blockchain
3. Wiped everything and installed 1.6.2 again and it does not seem to sync
4. I restarted already multiple times after waiting for a couple of hours and nothing happend

Cheers

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.6.2
chainnetwork: bitcoin / main
 17:33:17 up 11:05,  2 users,  load average: 0.67, 0.44, 0.27

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-12-10 06:31:11 GMT; 11h ago
  Process: 4935 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 4944 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 5014 (bitcoind)
    Tasks: 17 (limit: 4915)
   Memory: 44.5M
   CGroup: /system.slice/bitcoind.service
           └─5014 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid

Dec 10 06:31:11 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 10 06:31:11 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Thu 2020-12-10 17:33:17 GMT. --
Dec 10 06:31:11 raspberrypi systemd[1]: Starting Bitcoin daemon...
Dec 10 06:31:11 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-12-10T15:06:26Z Potential stale tip detected, will try using extra outbound peer (last tip update: 30870 seconds ago)
2020-12-10T15:15:31Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: general failure
2020-12-10T15:16:56Z Potential stale tip detected, will try using extra outbound peer (last tip update: 31500 seconds ago)
2020-12-10T15:26:50Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: general failure
2020-12-10T15:27:26Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32130 seconds ago)
2020-12-10T15:37:56Z Potential stale tip detected, will try using extra outbound peer (last tip update: 32760 seconds ago)
2020-12-10T15:48:26Z Potential stale tip detected, will try using extra outbound peer (last tip update: 33390 seconds ago)
2020-12-10T15:51:27Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: general failure
2020-12-10T15:58:56Z Potential stale tip detected, will try using extra outbound peer (last tip update: 34020 seconds ago)
2020-12-10T16:02:05Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: general failure
2020-12-10T16:09:26Z Potential stale tip detected, will try using extra outbound peer (last tip update: 34650 seconds ago)
2020-12-10T16:18:11Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: general failure
2020-12-10T16:19:56Z Potential stale tip detected, will try using extra outbound peer (last tip update: 35280 seconds ago)
2020-12-10T16:30:26Z Potential stale tip detected, will try using extra outbound peer (last tip update: 35910 seconds ago)
2020-12-10T16:40:56Z Potential stale tip detected, will try using extra outbound peer (last tip update: 36540 seconds ago)
2020-12-10T16:51:26Z Potential stale tip detected, will try using extra outbound peer (last tip update: 37170 seconds ago)
2020-12-10T16:56:30Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: general failure
2020-12-10T17:01:56Z Potential stale tip detected, will try using extra outbound peer (last tip update: 37800 seconds ago)
2020-12-10T17:12:26Z Potential stale tip detected, will try using extra outbound peer (last tip update: 38430 seconds ago)
2020-12-10T17:22:56Z Potential stale tip detected, will try using extra outbound peer (last tip update: 39060 seconds ago)

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-12-10 06:31:11 GMT; 11h ago
  Process: 5015 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 5017 (lnd)
[... 95 lines cut ...]
localIP='192.168.178.37'
tempCelsius='49.6'
uptime=39910
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=0
initialSync=1
syncProgress=0.00
startcountLightning=1
lndActive=1
walletLocked=0
syncedToChain=0
scanTimestamp=1231006505
scanDate='Sat  3 Jan 18:15:05 GMT 2009'
scanProgress=.01
lndRPCReady=1
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

rootzoll, 2020-12-10:
Please use `CTRL+c`to get to ther terminal.

Can you confirm that the result of the command `admin@Panikorchester:~ $ bitcoin-cli getnetworkinfo | grep "connections"` is `"connections": 0`?

If so please run command `bitcoin-cli addnode "143.110.240.87:8333" "onetry"` wait a moment and does that change the result of the connections?

You can get back to checking the sync with the command `raspiblitz` then.

JGalt040, 2020-12-10:
Thanks for your feedback! I followed the instructions. If I check for the connections at first it gives a "1" as the answer, but if I check like a minute later it gives again "0" as answer.

I tried this after seeing no syncing process after ~1.5h.

rootzoll, 2020-12-10:
OK maybe you just need to find a stable node to connect to to get started ... maybe try to add with the `bitcoin-cli addnode` command 5 to 10 nodes (use a diverse mix of IPv4 & Tor) from https://bitnodes.io/nodes/ and see if it catches up.

rootzoll, 2020-12-11:
TODO: Remove the added TOR peers from bitcoin.conf and add peer dynamically when detecting 0 peers

JGalt040, 2020-12-11:
Thanks Christian, it is working now!

closed 2020-12-11 by JGalt040

woeisme, 2023-02-06:
I've noticed this issue with my node today, failed all connections to (old version?) tor nodes. I noticed they're using shorter length names than TorV3 supports. Might that be the problem?

Solved by adding a few nodes manually. Still took a while to figure out what was wrong.