raspiblitz/raspiblitz #1861: Issue recovering LND wallet

jayrmotta, 2020-12-11:
# Context

I've deposited funds in an old version of Raspiblitz and when I tried restoring my wallet after updating the node I never managed to access my funds anymore.

Back then there was no channel backup, but the funds were on chain (not committed to any channel) anyway.

Recently I've update to 1.6.2 and entered my mnemonic from the old node and this is what I see while trying to recover:

[image]
[image]

I imagine the chances of recovering my funds are low by now, but if there is still a chance I'd like to try, if someone could point to a path that depends less on raspiblitz software and more on the underlying software so I can try faster.

openoms, 2020-12-11:
try exiting this screen and use the lnd cli:

`lncli create`

will offer you to enter the seed and will restore your onchain funds.
Can also use the LND Lighning Desktop App as in the FAQ.

jayrmotta, 2020-12-12:
I've managed to recover my funds last evening, I wasn't entering the password D since the message suggests most people don't set that password, but apparently I did.

I recovered it using the Zap wallet though.

closed 2020-12-12 by jayrmotta

chGoodchild, 2024-04-14:
> try exiting this screen and use the lnd cli:
> 
> `lncli create`
> 
> will offer you to enter the seed and will restore your onchain funds. Can also use the LND Lighning Desktop App as in the FAQ.

The [LND Lightning Desktop App](https://github.com/lightninglabs/lightning-app/releases) was archived in 2021. Do you still recommend it or is there a better option now @openoms ?

The LND Lightning Desktop App says "loading network" since quite a while after accepting the seed phrase:
[image]

openoms, 2024-04-15:
> > try exiting this screen and use the lnd cli:
> > `lncli create`
> > will offer you to enter the seed and will restore your onchain funds. Can also use the LND Lighning Desktop App as in the FAQ.
> 
> The [LND Lightning Desktop App](https://github.com/lightninglabs/lightning-app/releases) was archived in 2021. Do you still recommend it or is there a better desktop option now @openoms ?
> 
> The LND Lightning Desktop App says "loading network" since quite a while after accepting the seed phrase: ![image](https://private-user-images.githubusercontent.com/1100745/322271051-0690e6ad-2565-4a70-a52f-410ea4f35697.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMxNjc5MDUsIm5iZiI6MTcxMzE2NzYwNSwicGF0aCI6Ii8xMTAwNzQ1LzMyMjI3MTA1MS0wNjkwZTZhZC0yNTY1LTRhNzAtYTUyZi00MTBlYTRmMzU2OTcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTVUMDc1MzI1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDQ4NTE0Yjk3YjNlZWE3Zjc1OWNjNjhhMmRkYjI5N2RkNmRiOTE3YTA2ZGI3ZjBjMDhlMzE4YmIyOWU0MjJjMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ATEZmQNj2w16hC3ffuHwqOGxQMbTZw4MDb-7m-mKTXo)

No, do not use that outdated app. 
Can recover an LND seed (and SCB) in Blixt wallet on mobile.

chGoodchild, 2024-05-04:
I didn't have any SCB and the funds weren't found in Blixt, so I used sparrow wallet to recover what was accessible on-chain. 

It looks like some funds might still be locked in private lightning channels. Is it possible to break these channels and free the funds by deliberately publishing an outdated state?