raspiblitz/raspiblitz #1862: Sphinx App Connection

rootzoll, 2020-12-11:
Go the following report on Telegram:

```
For people having problem connecting with Raspiblitz, I saw some cases where Relay finished the setup but Sphinx doesn't know about it, might have been because of slow connection or Tor problems, I'm not sure.

The solution, if you're not worried about your Sphinx DB which you shouldn't because you never used it, is to run the following commands

sudo mv /mnt/hdd/app-data/sphinxrelay/sphinx.db /mnt/hdd/app-data/sphinxrelay/old-sphinx.db

sudo systemctl restart sphinxrelay

That will move the sphinx db to a new file so it gets recreated when restarted

After this a new connection string will be created so you need to do the following to see it again:

/home/admin/config.scripts/bonus.sphinxrelay.sh status
If that doesn’t work for you, let me know
```

My first guess: Could it be that the HTTP request to the Relay is kind of cached and because its not working on first try (when relay is not ready) it also does not work on following calls (because the caches request result is send)? That would explain why a reset of the DB might fix this, because it creates a new secret string and the next request from the App will be done with a fresh URL that was never cached before.

The question on this might then be: "Where does the Chaching happen?"
- On the Sphinx App side?
- On the IP2Tor somehow?
- On the nginx reverse HTTPS Proxy?
- On the Sphinx Relay Server?

@frennkie can we rule out any chaching on the IP2Tor & ngnix sinde?

gonzaloaune, 2020-12-11:
This might be a race condition as well or the iOS app (not sure about Android) not waiting enough for the request to finish. I've seen this happening before where:

- iOS App sends request
- Relay takes too long to finish the request and sends back the info
- iOS App already timed out and assumed the setup process failed

Now iOS App and Relay know 2 different things, I talked to Tom (the iOS app dev) about this issue and we found a way for the mobile apps to at least retry in that case so this might not be an issue in the future. I think this is purely a communication problem between Mobile <> Relay

rootzoll, 2020-12-15:
Make sure to have and install the latest Sphinx Relay Server for v1.6.3 release.

rootzoll, 2020-12-18:
OK Sphinx is always installing the latest master from server - so no need for an update at the moment:
https://github.com/rootzoll/raspiblitz/blob/9210987ebba272780eb538c15c5fcef6f5abecc8/home.admin/config.scripts/bonus.sphinxrelay.sh#L321

The Sphinx Relay is now writing its standard out to the journal service and is listed on debug out.
Should be fine for final testing.

The34Goon, 2021-01-11:
Trying to update my IP2TOR subscription for Sphinx I get this error:

### Exception on Subscription

Unknown Error happened - please report to developers: ('letsancrypt acme failed', "# preparing DUCKDNS 
 creating certs for XXXX.XXXX.XXX
[Mon 11 Jan 20:46:50 CET 2021] Creating domain key 
[Mon 11 Jan 20:46:50 CET 2021] The domain key is here: /mnt/hdd/app-data/letsencrypt/certs/XXXX.XXXX.XXX_ecc/XXXX.XXXX.XXX.key
[Mon 11 Jan 20:46:50 CET 2021] Single domain='XXXX.XXXX.XXX'
[Mon 11 Jan 20:46:50 CET 2021] Getting domain auth token for each domain
[Mon 11 Jan 20:46:53 CET 2021] Getting webroot for domain='XXXX.XXXX.XXX'
[Mon 11 Jan 20:46:53 CET 2021] Adding txt value: XXXXXXXXXYYYYYYYYYYYYYZZZZZZZZZ for domain:  _acme-challenge.XXXX.XXXX.XXX
[Mon 11 Jan 20:46:53 CET 2021] Trying to add TXT record
[Mon 11 Jan 20:46:54 CET 2021] Errors happened during adding the TXT record, response=KO
[Mon 11 Jan 20:46:54 CET 2021] Error add txt for domain:_acme-challenge.XXXX.XXXX.XXX
[Mon 11 Jan 20:46:54 CET 2021] Please add '--debug' or '--log' to check more details.
[Mon 11 Jan 20:46:54 CET 2021] See: https://github.com/acmesh-official/acme.sh/wiki/How-to-debug-acme.sh
ls: cannot access '/mnt/hdd/app-data/letsencrypt/certs/XXXX.XXXX.XXX_ecc//fullchain.cer': No such file or directory
error='acme failed'
")

rootzoll, 2021-01-12:
@frennkie can you a reason why the acme script fails to work of accessing the cert?

m00ninite, 2021-03-15:
Any idea what the deal is here? I'm able to get Sphinx to connect briefly, but then it just ends up stuck trying to reconnect to LND.

> admin@raspberrypi:~ $ sudo journalctl -u sphinxrelay
-- Logs begin at Sun 2021-03-14 22:14:55 GMT, end at Mon 2021-03-15 03:59:57 GMT. --
Mar 14 22:20:39 raspberrypi systemd[1]: Starting SphinxRelay...
Mar 14 22:20:46 raspberrypi bonus.sphinxrelay.sh[3826]: # ok - copied fresh config.json & app.json into dist directory
Mar 14 22:20:46 raspberrypi systemd[1]: Started SphinxRelay.
Mar 14 22:20:54 raspberrypi env[4295]: => env: production
Mar 14 22:20:54 raspberrypi env[4295]: => process.env.PORT: undefined
Mar 14 22:20:54 raspberrypi env[4295]: => config.node_http_port: 3300
Mar 14 22:20:54 raspberrypi env[4295]: => [db] starting setup...
Mar 14 22:20:55 raspberrypi env[4295]: => [db] done syncing
Mar 14 22:20:55 raspberrypi env[4295]: => [db] setup done
Mar 14 22:20:55 raspberrypi env[4295]: => 2021-03-14 22:20:55 [lnd] reconnecting... attempt #1
Mar 14 22:20:55 raspberrypi env[4295]: Node listening on 3300.
Mar 14 22:20:55 raspberrypi env[4295]: (node:4295) DeprecationWarning: grpc.load: Use the @grpc/proto-loader module with grpc.loadPackageDefinition instead
Mar 14 22:20:55 raspberrypi env[4295]: (Use `node --trace-deprecation ...` to show where the warning was created)
Mar 14 22:20:57 raspberrypi env[4295]: => 2021-03-14 22:20:57 [lnd] reconnecting... attempt #2
Mar 14 22:20:58 raspberrypi env[4295]: 14 UNAVAILABLE: failed to connect to all addresses
Mar 14 22:20:59 raspberrypi env[4295]: => 2021-03-14 22:20:59 [lnd] reconnecting... attempt #3
Mar 14 22:21:01 raspberrypi env[4295]: => 2021-03-14 22:21:01 [lnd] reconnecting... attempt #4
Mar 14 22:21:03 raspberrypi env[4295]: => 2021-03-14 22:21:03 [lnd] reconnecting... attempt #5

gonzaloaune, 2021-03-15:
Are you running the latest version of Sphinx on your phone and also the latest version of Relay? @m00ninite

m00ninite, 2021-03-15:
@gonzaloaune The app is up to the latest on iOS app store.. the relay, I'm not sure. I just installed it this weekend via the raspiblitz menu. If it's grabbing the latest code off github, I'd assume it's the latest, or at least very recent.

m00ninite, 2021-03-18:
Ok, I ended up using raspiblitz's built-in update mechanism for sphinx-relay and it fixed it. Go figure :)

rootzoll, 2021-03-22:
OK for the next release RaspiBlitz will now check if the connection string has already been used and recommends actions to act on (export keys, deinstall/reinstall to also get to latest release tag version & updating server URL). This might not fix every issue repoerted here - but please if you find further issues on testing upcomming 1.7RC1 and above please open a fresh issue so that we can keep a better overview. 

Closing this issue and will do further final testing on #2089

closed 2021-03-22 by rootzoll