raspiblitz/raspiblitz #1864: Trouble accessing BTRFS feature during initial setup

Peter-Bravestrong, 2020-12-12:
I have the 1TB SanDisk SSD Plus plugged into one of the USB 3 ports, and an additional thumb drive plugged into the other USB 3 port. Setup says "hard disk too small" and to start over if I'd like to use a larger hard drive. Setup is basically ignoring the 1TB drive, finding the thumb drive instead, and reporting it as too small (which it is of course). When I run setup with 1TB drive only--no problem, but no BTRFS option. Any idea why setup ignores 1TB drive when thumb drive is also plugged in? (I'd really like to try the BTRFS option.)

rootzoll, 2020-12-14:
Is this with the latest v1.6.2 version? How big is your thumb drive? At the moment the process is designed to use a thumbdrive of max 32GB or to add it later.

Idea for v1.7 release: Make detection always to sort devices by size.

Peter-Bravestrong, 2020-12-15:
Ok, that explains it. My thumb drive is 64GB. Thanks.

closed 2020-12-15 by Peter-Bravestrong