raspiblitz/raspiblitz #1865: Best way to switch to a different drive

rumesht, 2020-12-13:
Hey,
my raspiblitz is running on a 2tb hdd because when setting it up i didn't have any other and that 2tb drive wasn't used.
now i would like to switch to a 0.5tb drive. 
is there a good way without downloading the blockchain again? 

i found the copystation script is that still working or should i do something else? 
Tank you

openoms, 2020-12-13:
don't switch to a 500GB drive, you won't be able to fit the blockchain and  the electrum database any more. 

Best way is to boot a linux (USB Ubuntu for example). Connect your drive and resize the disk (partition).
Once resized to equal or smaller as your new disk clone it with Linux with both disks connected to the Linux PC. 

If this sounds too technical create a migration file form the REPAIR menu, download it and start a with  a new SDcard and the new disk attached. 

Will need to redownload the blocks or copy the from the old disk with Linux.

Using an 1TB SSD is recommended as in the shopping list.

rumesht, 2021-01-22:
Thank you for your comment.
Just for people finding this thread searching for the same anwer:

I ended up deleting my raspiblitz data apart from the blockchain. Then booted a usb ubuntu as told above.
With gparted it was easy to resize the 2TB partition into 3xxGB(just the blockchain). I then copied it to my 500GB drive and resized the partition again to use the full 5xxGB drive.
I dont use electrum so that was enough for just the bitcoin core node + lnd.

Works like a charm now again without needing to download the blockchain.

closed 2021-01-24 by rootzoll