raspiblitz/raspiblitz #1866: BTCpayServer update

openoms, 2020-12-13:
Latest release 1.0.6.3

openoms, 2020-12-13:
Having some issues starting:
```
$ sudo journalctl -fu  btcpayserver.service 
-- Logs begin at Sun 2020-12-13 12:11:46 GMT. --
Dec 13 20:06:29 raspberrypi dotnet[14003]: info: Configuration:  Data Directory: /home/btcpay/.btcpayserver/Main
Dec 13 20:06:29 raspberrypi dotnet[14003]: info: Configuration:  Configuration File: /home/btcpay/.btcpayserver/Main/settings.config
Dec 13 20:06:29 raspberrypi dotnet[14003]: info: Configuration:  Network: Mainnet
Dec 13 20:06:29 raspberrypi dotnet[14003]: info: Configuration:  Supported chains: BTC
Dec 13 20:06:31 raspberrypi dotnet[14003]: info: BTCPayServer.Plugins.PluginManager: Loading plugins from /home/btcpay/.btcpayserver/Plugins
Dec 13 20:06:33 raspberrypi dotnet[14003]: info: BTCPayServer.Plugins.PluginManager: Adding and executing plugin BTCPayServer - 1.0.6.3
Dec 13 20:06:33 raspberrypi dotnet[14003]: info: Configuration:  Network: Mainnet
Dec 13 20:06:33 raspberrypi dotnet[14003]: info: Configuration:  Supported chains: BTC
Dec 13 20:06:33 raspberrypi dotnet[14003]: fail: Configuration:  No database option was configured.
Dec 13 20:06:33 raspberrypi systemd[1]: btcpayserver.service: Succeeded.
```

rootzoll, 2020-12-15:
Merged PR. Put on final testing.

lets-fuckin-do-it, 2020-12-16:
Who hooo you guys are rock stars, I was thinking on Monday "dohhh I really need the new Webhooks features of the new BTCPay" and here you are final-testing it. Looking forward to it :-)

rootzoll, 2020-12-24:
OK install works - will be part of v1.6.3 release - closing issue.

closed 2020-12-24 by rootzoll