raspiblitz/raspiblitz #1867: [64 bit] Testing the minimal pure Debian image

openoms, 2020-12-13:
Discussion from: https://github.com/rootzoll/raspiblitz/issues/1199#issuecomment-742739629

Truly minimal image: https://raspi.debian.net/

Issues so far:
* can't ssh in on first boot. Need to log in with a HDMI screen, create and new user to allow ssh. (https://raspi.debian.net/defaults-and-settings/)
* no `raspi-config` or `locale-gen` available to install (no readme in: https://github.com/RPi-Distro/raspi-config)

openoms, 2021-02-08:
PGP verification:

$ cat 20201112_raspi_4.xz.sha256.asc 
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

56d6e5c674fb89be07ed160807d0b166cc3713d8c32c47ff61fbc94f39452373  20201112_raspi_4.img.xz
-----BEGIN PGP SIGNATURE-----

iHUEARYIAB0WIQRgswk9lhCOXLlxQu/i9jtDU/RZiQUCX61zcwAKCRDi9jtDU/RZ
ieP+AQCpUvxsiswIpHTRzxw1/3QlfI4wxgL8BVMixwp/37xBOQD/eDzqxyRVhkjX
ywV0nAAEMJestu2TQHAufqPBfwCrxQY=
=s/jY
-----END PGP SIGNATURE-----

$ sha256sum 20201112_raspi_4.img.xz 
56d6e5c674fb89be07ed160807d0b166cc3713d8c32c47ff61fbc94f39452373  20201112_raspi_4.img.xz

gpg --receive-key 60B3093D96108E5CB97142EFE2F63B4353F45989
gpg: DBG: locking for '/home/s/.gnupg/pubring.kbx.lock' done via O_EXCL
gpg: key 2404C9546E145360: public key "Gunnar Wolf <gwolf@gwolf.org>" imported
gpg: Total number processed: 1
gpg:               imported: 1

gpg --verify 20201112_raspi_4.xz.sha256.asc 
gpg: Signature made Thu 12 Nov 2020 17:40:03 GMT
gpg:                using EDDSA key 60B3093D96108E5CB97142EFE2F63B4353F45989
gpg: Good signature from "Gunnar Wolf <gwolf@gwolf.org>" [unknown]
gpg:                 aka "Gunnar Eyal Wolf Iszaevich <gwolf@iiec.unam.mx>" [unknown]
gpg:                 aka "Gunnar Wolf <gwolf@debian.org>" [unknown]
gpg: Note: This key has expired!
Primary key fingerprint: 4D14 0506 53A4 02D7 3687  049D 2404 C954 6E14 5360
     Subkey fingerprint: 60B3 093D 9610 8E5C B971  42EF E2F6 3B43 53F4 5989
gpg: WARNING: not a detached signature; file '20201112_raspi_4.xz.sha256' was NOT verified!
```

openoms, 2021-02-08:
Kernel on first boot:
```
Linux rpi 5.8.0-0.bpo.2-arm64 #1 SMP Debian 5.8.10-1~bpo10+1 (2020-09-26) aarch64 GNU/Linux
```

after `apt upgrade` and `reboot`:
```
Linux rpi 5.9.0-0.bpo.5-arm64 #1 SMP Debian 5.9.15-1~bpo10+1 (2020-12-31) aarch64 GNU/Linux
```

openoms, 2021-02-08:
The incident described here: https://www.reddit.com/r/linux/comments/lbu0t1/microsoft_repo_installed_on_all_raspberry_pis/
and dealt with in:
https://github.com/rootzoll/raspiblitz/pull/1858#issuecomment-773307781
sparked my interest again to setup on the Pure Debian Image.
Manage to iron out some initial difficulties already, see as edited the comment above.

openoms, 2021-02-11:
Documenting the SDcard release process for the Joininbox based on the pure Debian image: https://github.com/openoms/joininbox/blob/master/FAQ.md#build-the-sdcard-image

openoms, 2021-02-12:
New release of the pure debian image: https://raspi.debian.net/tested-images/
* Image: https://raspi.debian.net/verified/20210210_raspi_4_buster.img.xz
* Signature: https://raspi.debian.net/verified/20210210_raspi_4_buster.xz.sha256.asc

openoms, 2021-02-17:
Will test building the SDcard image with this PR: https://github.com/rootzoll/raspiblitz/pull/1858
Trying to find out how to identify the pure Debian image vs Armbian.
For now adding the `raspi-config` manually, but in fact it can be replaced with the Debian scripts, although it might come handy for setting up wifi and other RPi specific settings (to be tested).
```
$ uname -n
rpi4-20210210

$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 10 (buster)"
NAME="Debian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

openoms, 2021-02-17:
So far did not get an output on the GPIO LCD despite installing the 64bit drivers.

The first comment about using the 64bit Raspberry OS as the base image for the raspiblitz image release based still stands for reasons:
* GPIO LCD (should not be difficult to make it work)
* need to install `lightdm` and `lxde` for the touchscreen for now which come with hundreds of adsed of packages during the SDcard build

xanoni, 2021-08-02:
Despite all the benefits (e.g., full RAM addressability), there seem to be certain disciplines where the 32bit Raspbian substantially outperforms a 64bit OS. Disclaimer: I haven't had time yet to confirm this myself with the latest OS images.

What was most worrisome to me is that 64bit may affect the feasibility of Full-Disk-Encryption (see our discussion here: https://github.com/rootzoll/raspiblitz/issues/1681), given that AES performance is substantially worse in benchmarks. I don't know if this would also affect other encryption ciphers, such as `xchacha12` / `xchacha20`. It could be that those suddenly perform better, who knows.

[image]

AES might also become a bottleneck in other areas, but surprisingly VPN seems fine. I assume they used a different encryption cipher.

[image]

More benchmarks:

[image]

[image]

**Sources:**
https://www.cnx-software.com/2020/01/29/raspberry-pi-4-benchmarked-with-32-bit-and-64-bit-debian-os/
https://hackaday.com/2020/01/28/raspberry-pi-4-benchmarks-32-vs-64-bits/

xanoni, 2021-08-10:
Very exciting news, @openoms @nyxnor !

While I still haven't switched to 64-bit myself and can't say if we have a solution, there are two very positive developments.

1) We may no longer need the AES performance for fast disk encryption. I was able to more than double the performance of the alternative, software-optimized `XChaCha12` and `XChaCha20` encryption ciphers by using a `4096B` sector size with my SSD (instead of the default/physical `512B` sectors). The speed is now far closer to an unencrypted drive in many real-world benchmarks, and up to ~45% faster than `AES-XTS`. The huge differences to unencrypted SSDs occur mostly in synthetic/unrealistic benchmarks (e.g., very long sequential reads).

See all the gory detail here: https://github.com/keks24/raspberry-pi-luks/issues/2#issuecomment-896347725

2) I found the code that the Kernel uses to emulate AES when no specialized hardware circuits are present (like in the Pi), and there have been many updates to the codebase since the benchmarks (see previous post) were created in early 2020. Thus, there could be a huge improvement by now, I need to test that! See the respective folder in [the Linux Kernel Repository ](https://github.com/torvalds/linux/tree/master/arch/arm64/crypto). The component we're interested in is called `NEON`.

xanoni, 2021-08-12:
My next step is to benchmark everything again on a 64-bit image ... I could just do this with an unencrypted root, but I want to do it properly, given you can't really trust SSDs if you ever put anything unencrypted on them. Also, in the end it has to be done anyway, so why not do it now.

Thus, I'm trying to recreate the full disk encryption playbook steps with the 64-bit image. It hasn't all gone smoothly so far, but I should be there very soon.

I'm documenting my progress here: https://github.com/keks24/raspberry-pi-luks/issues/12

xanoni, 2021-08-13:
Haven't had time to continue here, but found some very promising (although synthetic) benchmarks ... https://www.raspberrypi.org/forums/viewtopic.php?p=1733736&sid=4d711bc22d3c89b267171771310f8f11#p1733736

Please note he is saying (shouldn't cause a huge delta):
```
my Pi is slightly overclocked and undervoltaged:
arm_freq=1600
over_voltage_delta=-5000
```

xanoni, 2021-08-13:
I got the full disk encryption working on 64-bit RaspiOS Lite and performance looks nice. Will do a proper comparison between 32-bit and 64-bit this weekend and share it here.

xanoni, 2021-08-14:
Okay, I think we're good. With the latest kernel, disk encryption is clearly faster with the 64-bit than with the 32-bit Kernel. And that's before tweaks.

**Same benchmark as above but with overclocked CPU (1500 MHz => 1800 MHz):**

```bash
# Tests are approximate using memory only (no storage IO).
#            Algorithm |       Key |      Encryption |      Decryption
xchacha12,aes-adiantum        256b       201.7 MiB/s       196.4 MiB/s
xchacha20,aes-adiantum        256b       161.2 MiB/s       169.1 MiB/s
aes-xts                       256b       100.3 MiB/s        88.7 MiB/s
```

**And here with 2100 Mhz:**

```bash
# Tests are approximate using memory only (no storage IO).
#            Algorithm |       Key |      Encryption |      Decryption
xchacha12,aes-adiantum        256b       225.3 MiB/s       224.1 MiB/s
xchacha20,aes-adiantum        256b       181.4 MiB/s       192.3 MiB/s
aes-xts                       256b       114.0 MiB/s       101.1 MiB/s
```

**2100 MHz seems to be stable on my PI, so I'm keeping that for now. Below the key settings.**

```bash
initial_turbo=60 # max: 60
over_voltage=6 # [-16,8] ### TODO: try lowering to 5
arm_freq=2100 # default: 1500 ### TODO: try 2150-2200
core_freq=550 # default: 500
```

rootzoll, 2021-08-16:
Are you already coordinating with @AaronDewes on testing a minimal pure Debian image .. I can remember he also had interest on this - see #2389

AaronDewes, 2021-08-16:
No, and I haven't really spent time on the minimal Debian recently, I'll improve it soon.

openoms, 2021-11-08:
There is a stable Debian 11 Bullseye version out: https://raspi.debian.net/tested-images/

cstenglein, 2022-02-13:
An alternative would be to use the Raspberry Pi OS Lite Version, so there is less difference to the currently used image + less packages (no desktop + stuff like libreoffice):

https://downloads.raspberrypi.org/raspios_lite_arm64/images/

openoms, 2022-02-13:
@cstenglein see our last attempt to use the Raspbian Lite here: https://github.com/rootzoll/raspiblitz/issues/1184

openoms, 2024-02-10:
closing this as stale. Note the https://github.com/openoms/joininbox is build using the image from https://raspi.debian.net for the RPi.

closed 2024-02-10 by openoms