raspiblitz/raspiblitz #1869: Trouble with Raspiblitz setup

duckwaffles, 2020-12-13:
I am having two issues.  First, I have tried two screens and they both only turn on and display white.  I setup the raspiblitz anyway via SSH.  However for 2 1/2 days it has been stuck at:

Node is Syncing (0)
Blockchain Progress: 0.00%
Lightning Progress: .01%

I checked the error log and it appears that it is not connecting.  I forwarded port 8333 and rebooted but it does not appear to have made a difference.

duckwaffles, 2020-12-14:
From the log:

2020-12-13T23:03:46Z Potential stale tip detected, will try using extra outbound                                                                                                              peer (last tip update: 1891 seconds ago)
2020-12-13T23:14:16Z Potential stale tip detected, will try using extra outbound                                                                                                              peer (last tip update: 2521 seconds ago)
2020-12-13T23:24:46Z Potential stale tip detected, will try using extra outbound                                                                                                              peer (last tip update: 3151 seconds ago)
2020-12-13T23:27:29Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: gen                                                                                                             eral failure
2020-12-13T23:35:16Z Potential stale tip detected, will try using extra outbound                                                                                                              peer (last tip update: 3781 seconds ago)
2020-12-13T23:45:46Z Potential stale tip detected, will try using extra outbound                                                                                                              peer (last tip update: 4411 seconds ago)
2020-12-13T23:47:20Z Socks5() connect to uz3pvdhie3372vxw.onion:8333 failed: gen

rootzoll, 2020-12-14:
See help on this issue here: https://github.com/rootzoll/raspiblitz/issues/1859#issuecomment-742695230

nyxnor, 2021-04-18:
Please close this issue if solved.

closed 2021-04-19 by rootzoll