raspiblitz/raspiblitz #1870: Unable to download blockchain from laptop on win7 via SCP

maaxxyy, 2020-12-14:
Hi as described above, it took 9 days to download the blockchain to my laptop (win7), now i am finding it near impossible to transfer the files via scp, I can ssh into my pi with out any issues. I have also installed winSCP and all i get is a permission error. 
I'll post some screen shots  shortly to explain my issue more in depth. Thanks

rootzoll, 2020-12-14:
Are you running an RaspberryPi3 or a 4? Are you using the latest v1.6.2 release?

maaxxyy, 2020-12-14:
hi i'm using pi 3 b+, using v1.6.1 release.

rootzoll, 2020-12-14:
I just tested again ... its a 2.3GB file that is a disk image:
https://github.com/rootzoll/raspiblitz#downloading-the-software

maaxxyy, 2020-12-14:
Ah right i was looking at the releases on the right hand of the front page, i'll download this and see how i get on, thanks.

maaxxyy, 2020-12-14:
Hi just flashed the new software to sd card, running into the same problem when trying to run - 
scp -r ./chainstate ./blocks bitcoin@192.168.0.51:/mnt/hdd/bitcoin
scp is not recognised as a internal or external command, operable 
I have putty, pscp.exe and winscp installed but not able to transfer the files.

rootzoll, 2020-12-14:
I dont have a WIndows 7 to test it with ... please let me know if that woked with "pscp -scp -r -P 22 ./chainstate ./blocks bitcoin@192.168.0.51:/mnt/hdd/bitcoin" would be good to add that info to the script.

maaxxyy, 2020-12-14:
Yes that worked, its downloading right now. I ssh into pi on my win7 laptop (where i have installed bitcoin core), then opened a new terminal and wrote -
pscp -scp -r -P 22 ./chainstate ./blocks bitcoin@192.168.0.51:/mnt/hdd/bitcoin

I also installed putty, winscp and pscp.exe to get to the above solution (you may not need to install winscp)
This will take a few hours to complete.

nyxnor, 2021-04-18:
@maaxxyy Good that is worked. Please clos this issue.

closed 2021-04-19 by rootzoll