raspiblitz/raspiblitz #1873: Backup Options

velgard1, 2020-12-15:
Hi Team,

I'm a little confused with all the backup options and want to know what's necessary / easy / useful with my current configuration.
I use 1.6.2 with BTRFS (64GB USB Stick) and have RTL, RPC Explorer and Lightning Loop enabled right now.

Somewhere it's mentioned that you shouldn't use sticks with more than 32GB since this could cause power issues - so I wonder if it's save to use another stick for backup because you can't use the one used for BTRFS.
On the other hand you can create a backup file to download and also use a function to backup within RTL...

I simply want to ensure that I can restore everything properly after an update or an hardware error and would appreciate advice.

Thanks

rootzoll, 2020-12-15:
The bigger and older the thumb drive is the higher the chance that it consumes more power. 64GB should be OK but switching down to a 32GB is best practice.

To change BTRFS RAID Thumbdrive .. you should be able to just turn that "off" and replace the thumb drive and switch it "on" again and your fine - use commands:
`sudo /home/admin/config.scripts/blitz.datadrive.sh raid off` &
`sudo /home/admin/config.scripts/blitz.datadrive.sh raid on`
the name of the raid device you should get from
`sudo /home/admin/config.scripts/blitz.datadrive.sh status`

Regarding backup files for download:

There is the migration file that makes a archive of all your `relevant data` .. so no blockchain data or indexes but all wallets and data from installed apps. You can use that to move to a complete new hardware - like a new HDD/SSD.

And there is the smaller brother the lnd-rescue file that just backups all the relevant data from your LND - but not the other apps.

velgard1, 2020-12-16:
Sorry to ask for clarification, but I'quite new to Linux / CLI...

So when I simply stop raid, change the stick & start raid again - why do I need this status command?

It seems lnd rescue is the preferred solution when I do normal updates just in case something goes wrong - but for my understanding I don't have to use the rescue file and when I update to a new version everything should work fine normally, right?

Thanks again

rootzoll, 2020-12-19:
> So when I simply stop raid, change the stick & start raid again - why do I need this status command?

When turning the raid on you have to give the name of the device (could be 'sdc` for example) .. the `status` command gives you a list of possible names.

rootzoll, 2020-12-19:
> It seems lnd rescue is the preferred solution when I do normal updates just in case something goes wrong - but for my understanding I don't have to use the rescue file and when I update to a new version everything should work fine normally, right?

Yes normally you dont need the rescue file on an update ... its just for the case something goes worng. Once your node is up and running after upadte I would even recommend to delete the reschue-file.

velgard1, 2020-12-21:
Ok, thank you for clarification.

closed 2020-12-21 by velgard1