raspiblitz/raspiblitz #1878: Sphinx installation error

senf42, 2020-12-16:
During the install process for Sphinx, it mentions an error about IP2Tor (which I do not have installed, I'm using reverse proxies to externally connect to services, which has mostly worked fine for me). 

After finishing the install, loading http://<IP>:3300 and https://<IP>:3301 both give the same error: "Cannot GET /"

I tried the commands in Issue #1862 thinking they might help, but I still see the same error message.

rootzoll, 2020-12-18:
> After finishing the install, loading http://(local IP):3300 and https://(local IP):3301 both give the same error: "Cannot GET /"

this is the correct answere of the sphinx relay when api get called without correct path.

> During the install process for Sphinx, it mentions an error about IP2Tor (which I do not have installed, I'm using reverse proxies to externally connect to services, which has mostly worked fine for me).

Can you tell what this error was? Does the same error comes when you open the "SPHINX" option in MAINMENU after install?

senf42, 2020-12-18:
The error is:

Your IP2TOR+LetsEncrypt may have problems:
Connection String not updated yet. Try again a bit later or check for errors.

Check if locally responding: http://(local IP):3300

Check if service is reachable over Tor:
(Tor address)

Back    Continue Anyway

The Tor address is not reachable either. 

The SPHINX option from main menu shows the screen after I choose "Continue Anyway" which gives instructions on how to download the Sphinx Chat app and provides the http://(local IP):3300 address again, and suggests adding an IP2TOR Bridge.

rootzoll, 2020-12-18:
I think I found the basic warning ... changed it to not show up after install (might fix it self after server started).

You can try this patch with MAINMENU > UPDATE > PATCH > PATCH-MENU > PATCH and then deinstall & reinstall sphinx relay again.

senf42, 2020-12-18:
I no longer get the IP2Tor error, but I do still only see "Cannot GET /" after patching, uninstalling, and reinstalling.

rootzoll, 2020-12-19:
The "Cannot GET /" is the correct response from the server. I already mentioned to the Sphinx Relay Server Team that another response on the default path would help prevent misunderstandings - but thats up to them.

senf42, 2020-12-19:
Ah, I understand now. Maybe add something to the Sphinx info from the main menu? Like:

Local server for test & debug: http://(local IP):3300 (You should see "Cannot GET /" from a browser)

Thanks for the update and explanation!

rootzoll, 2020-12-21:
Added info about the correct server response. Closing this issue for v1.6.3 release.

closed 2020-12-21 by rootzoll